asrob-uc3m/impresoras-asrob #58: Nombre para la TronXY

Siotma, 2018-04-15:
Necesitamos bautizar nuestra impresora, ya que quedó pendiente de #51 .

De momento he recopilado las ideas que se han ido dando:

- TronXY (el más original).
- StarKiller (nombre del planeta que destrulle mundos en Star Wars VII).
- DeathStar (la Estrella de la Muerte dos veces destruida).
- Geonosis (planeta donde se fabrican droides de combate en el que C3PO dice la famosa frase de "robots fabricando robots, qué obscenidad").

Se aceptan todo tipo de sugerencias y comentarios.

Propongo seguir el siguiente esquema:
- [ ] Lluvia de ideas hasta el próximo domingo.
- [ ] Votación.

David-Estevez, 2018-04-15:
Caliprinter (porque se parece a un cubo de calibración)

JorFru, 2018-04-16:
Sugiero el nombre "Borg"
Se que es un nombre feo, pero tiene su explicacion.
Cubo Borg es el nombre de el transporte interestelar de los aliens Borg en la saga Star Trek.

[image]

GioSunrider, 2018-04-16:
Ya que te pones ponle C-Borg, que hasta si lo pronuncias parece un ciborg xD

JorFru, 2018-04-16:
C-Borg, definitivo

Siotma, 2018-04-16:
XY-Borg?

Actualizo la lista..

David-Estevez, 2018-04-16:
[image]

:joy: :joy: :joy:

GioSunrider, 2018-04-16:
Y, Mjölnir!? Como el martillo de Thor Dios del trueno! Porque nadie la va a poder levantar más que los signos de ello jajajajaajajaj

Siotma, 2018-04-21:
Hoy es el último día para aportar ideas, mañana abriré votación con los nombres que haya.

Siotma, 2018-04-22:
Votación abierta hasta el próximo jueves en https://goo.gl/forms/p1yVrF4RrPuYFcl52

Siotma, 2018-04-26:
Acabo de cerrar la votación. De 8 votantes, Geonosis ha sacado 4 votos, más que ninguna otra. Los resultados se pueden ver pulsando el enlace del form: https://goo.gl/forms/p1yVrF4RrPuYFcl52 .

TronXY: 3 votos.
StarKiller: 2 votos.
DeathStar: 1 voto.
**Geonosis: 4 votos.**
Caliprinter: 2 votos.
Borg: 1 voto.
C-Borg: 3 votos.
XY-Borg: 0 votos.
Mjölnir: 1 voto.
El que queráis, pesados: 1 voto.

[image]

Adorad a Geonosis!

closed 2018-04-26 by Siotma