asrob-uc3m/impresoras-asrob #59: Extrusor de pellets

dmarcosm, 2018-04-18:
Hago la propuesta de hacernos un extrusor de pellets. Dicho extrusor funcionaría usando restos de pla (que tenemos MUCHOS restos), y usaría un tornillo sinfín para empujar los pellets por un barrel/cuerpo modificado (más ancho). En el armario hay algunos materiales que podríamos usar para este proyecto si nos animamos a llevarlo a cabo. 
Ideas, sugerencias y opiniones son agradecidas :+1:

GioSunrider, 2018-04-18:
A mí como idea me mola, y a ver cuando montemos la granja me ofrezco como tributo para la ayuda jajjaajaj

JorFru, 2018-04-18:
Le veo futuro a la idea, me uno!
Podemos tomar ideas de este proyecto:
mahor.xyz

rsre, 2018-04-18:
Tengo entendido que el PLA se degrada con cada ciclo de vida. Un paper sobre el asunto: https://www.google.es/url?sa=t&source=web&rct=j&url=https://sffsymposium.engr.utexas.edu/sites/default/files/2017/Manuscripts/InvestigatingMaterialDegradationthroughtheRec.pdf&ved=2ahUKEwjfzsH06MTaAhVItxQKHf2RBoUQFjABegQIBBAB&usg=AOvVaw27aYQysHtZMiSubG673Zo7

Joder, en el móvil es imposible copiar el enlace de verdad.

GioSunrider, 2018-04-18:
Si, se degrada, obviamente no vamos a conseguir grandes piezas pero podemos cubrir muchas que no necesitamos que queden bien del todo. Y también tenemos mucho PLA roto antes del uso así que en ese sentido vamos bien.

dmarcosm, 2018-04-18:
> Le veo futuro a la idea, me uno!
Podemos tomar ideas de este proyecto:
mahor.xyz

De hecho lo que tenía en mente es casi lo mismo que ese proyecto!

> Tengo entendido que el PLA se degrada con cada ciclo de vida.

Yo pensaba mas de cara a usar filamentos rotos que tenemos como dice Jorge. También a usar pellets comprados directamente (el kg está a unos... 8/9€). Y sobre todo... por el progreso! Todo es probar :D

David-Estevez, 2018-04-19:
Tenéis a alguien que hizo esto mismo: Diego Trapero. Quizá él os pueda aconsejar sobre la viabilidad de este proyecto basado en sus experiencias.

jgvictores, 2018-04-19:
cc @diegotrap : )

diegotrap, 2018-04-19:
Sí, mi trabajo de fin de grado fue un extrusor de pellets, aún lo tengo. Creo que os referís a un extrusor para impresora 3D, no de filamento, pero os cuento mi experiencia igual.

Todos los archivos los publiqué: https://github.com/diegotrap/filament_extruder. Hay sketches de Arduino, modelos 3D, cuadernos de iPython... Los resultados están en mi TFG: https://e-archivo.uc3m.es/handle/10016/20112 (sorry por los 77Mb).

Un extrusor de filamento doméstico económicamente no tiene sentido y es muy dificil de hacer que produzca filamento con diámetro consistente. Para aprender sí que puede valer si tienes interés.

Un extrusor tipo Mahor sí que puede funcionar. Económicamente tampoco es un ahorro pero para probar a reciclar, para usar materiales raros y para aprender sí que vale. No sé como funcionará, Miguel Angel Salmerón [dice que va muy bien](https://www.youtube.com/watch?v=OsXVroP8Dg4).

dmarcosm, 2018-10-15:
Cierro por no haber ganas de hacerlo y por tener proyectos más interesantes a la vista! Si eso reabrimos en el futuro :)

closed 2018-10-15 by dmarcosm