asrob-uc3m/impresoras-asrob #60: Compras para la granja de impresoras

Siotma, 2018-04-21:
En la última reunión identificamos algunas cosas que hay que comprar para la granja.

La idea es ir proponiéndolas por esta Issue para que no se nos olviden.

Si alguien compra algo, primero debe quedar acordado aquí, no vaya a ser que lo compremos por duplicado. Recordad que hay que pedir factura con la siguiente información:

> Universidad Carlos III
Dpto. Ingeniería de Sistemas y Automática
Avda. de la Universidad 30
Leganés 28911
CIF: Q2818029G

- [ ] Regleta de 5 metros con mínimo 8 tomas de corriente, e interruptor. (Para las impresoras, el router y las raspberrys).
- [ ] Regleta de 3 metros con mínimo 5 tomas de corriente e interruptor. (Para poder poner hasta 2 PCs, 2 pantallas y la Arcade).
- Se aceptan propuestas hasta mañana lunes a las 23:59.

David-Estevez, 2018-04-21:
Ojo, si las compras son a china, los que llevan las cuentas prefieren que no se facture a nombre de la Universidad, sino a nombre de la asociación (que va a estar complicado) o a nombre de la persona que lo compra. Al memos eso nos dijeron la última vez...

Siotma, 2018-04-21:
Exacto! No lo he puesto porque en mi cabeza todas las compras necesarias se pueden hacer en una ferretería de barrio, pero en caso de que lleguemos a pensar en aliexpress o similar hay que tener eso muy en cuenta!

GioSunrider, 2018-04-23:
Y unos interruptores GPIO para poder apagar las impresoras a través de las RPis?

JorFru, 2018-04-24:
Anoto que la TronXY necesita un MOSFET para la cama, el suyo se pone horroroso de caliente. Es una mejora que añadirá seguridad a nuestra granja. Cuestan en torno a 8€. También se podría mirar comprar dos, por si surge una nueva impresora con la misma electrónica mierder.

Siotma, 2018-04-26:
El tema de los interruptores GPIO no me convence. De momento vamos a intentar montar la granja y eso vendrá en el futuro, una vez todo funcione.

Lo que sí me parece urgente es lo del MOSFET. Lo añado.

Seguid comentando!

Siotma, 2018-04-29:
Me he dado cuenta de que esta issue es exactamente eso que digo que no se debe hacer, citándome a mi mismo:

> Issues con tareas abstractas o muy abiertas no son eficientes porque siempre aparecen nuevas cosas que hacer y parece que es impsible cerrarlas. Además, este tipo de issues, así como las ideas que se dan en ellas, se pierden con el tiempo...

Declarada inválida.

closed 2018-04-29 by Siotma