asrob-uc3m/impresoras-asrob #61: Puesta a punto de TronXY (nombre provisional)

JorFru, 2018-04-24:
Abro issue para comentar las mejoras que se le harán a esta máquina.

Problema detectado: los cables del hot-end se calientan, no mucho pero se hacen notar, a lo largo de todo el cable. Esto seguramente sea por la baja calidad del cableado y su sección escasa. Habría que recablear con cable siliconado, hay stock en el armario, en la zona "maraña de cables".

- [ ] Recablear hot-end

dmarcosm, 2018-04-25:
Otra mejora que se le hará es imprimir los soportes de los motores de X e Y: https://www.thingiverse.com/thing:2613150
Otra cosa interesante que deberíamos imprimir es un mejor ventilador de capa. Actualmente gran parte del aire va casi directo al extrusor: https://www.thingiverse.com/thing:2706922/#files
Por último, hay que ajustar las tensiones de las correas y valorar si queremos hacer algún cambio a los rodamientos de las correas, ya que están como torcidos y hacen que estén moviendose arriba y abajo todo el rato.

dmarcosm, 2018-04-25:
Se me olvidó comentar el tema de la pantalla. Quizá no sea necesario al ponerla en la estantería. Pero si se va a quedar a la altura a la que está ahora, sería interesante mover la pantalla a la esquina superior derecha de la estructura, ya que tan abajo no se ve nada.

JorFru, 2018-04-25:
Para lo que dice David sobre la pantalla, habría que comprobar si los
cables llegan o en su caso se pueden empalmar.
Yo sugiero imprimir una chorripieza para colocar la pantalla en ángulo.

dmarcosm, 2018-04-27:
> Hay que ver también porque ayer cuando fuimos a imprimir, en la cuarta capa, se desplaza completamente. @anacarnerero y yo probamos a tensar las correas pero seguida el mismo problema. Hay que investigar el porque, ya que en ambos gcodes realiza exactamente el mismo desplazamiento.

A qué velocidad se estaban imprimiendo esos gcode?

Siotma, 2018-04-27:
Respecto a la pantalla, me gusta más la idea en ángulo, pero abajo, que subirla arriba, ya que estorbaría el campo de visión hacia la maravilla que es esta impesora imprimiendo.

dmarcosm, 2018-04-27:
> Hay que ver también porque ayer cuando fuimos a imprimir, en la cuarta capa, se desplaza completamente. @anacarnerero y yo probamos a tensar las correas pero seguida el mismo problema. Hay que investigar el porque, ya que en ambos gcodes realiza exactamente el mismo desplazamiento.

Me acabo de acordar que uno de los primeros días que imprimimos pasó lo mismo, y llegué a la conclusión de que era culpa de sobreextrusión. De vez en cuando dejaba pegotes, y si da la casualidad de que cruza por encima de ese pegote, intenta llevarselo por delante y pierde pasos. Quizá puede ir por ahí el problema.

Siotma, 2018-04-28:
> De vez en cuando dejaba pegotes, y si da la casualidad de que cruza por encima de ese pegote, intenta llevarselo por delante y pierde pasos.

Eso es un problema muy habitual con cualquier impresora. Habría que calibrar los pasos/mm del extrusor y seguir perfeccionando los perfiles, sobre todo jugar con la retracción...

David-Estevez, 2018-04-28:
Lo extraño es que siempre la impresión de estropee en el mismo punto exacto... Tiene más pinta de ser cosa del gcode...

dmarcosm, 2018-04-28:
Yo creo que justo por eso @daeste. Va dejando pegotes, pero no siempre vuelve a pasar por ellos de forma que quiera atravesarlos.
Entonces digamos que en la linea 645 de gcode deja el pegote, y en la 756 lo intenta atravesar. Eso va a pasar siempre que se use ese gcode, dejando el desplazamiento siempre en la misma altura.

David-Estevez, 2018-04-29:
Pero la pérdida de pasos por pegotes, que también la he sufrido alguna vez en Dédalo, hace que se pierdan pasos en distintas capas y que no siempre se pierda la misma cantidad de pasos. En mi experiencia, no suele ser tan repetible.

Siotma, 2018-05-06:
En la reunión del pasado jueves, Geonosis no pudo imprimir ninguna pieza al dar sucesivos errores de Thermal Runaway del extrusor. Se sospecha que el problema es que el ventilador de capa echa aire directamente sobre el hotend.

Propongo:
- Ejecutar el [PID Tuning](http://reprap.org/wiki/G-code/es#M303:_Run_PID_tuning) para asegurarnos de que el regulador PID de temperatura está bien ajustado.
- Taponar parcialmente la salida del aire del ventilador de capa para ver si el problema desaparece.

En [este manual](http://nathan.vertile.com/blog/2018/01/11/tronxy-x5s-build-guide/#x-and-y-carriage) dice:

> [...] I suggest printing this cooling system and installing it now: https://www.thingiverse.com/thing:2706922. You'll also need this adapter plate, if the included flan is a blower type: https://www.thingiverse.com/thing:2765424. The standardshroud will work OK as well, but is much less convenient, as the metal case covers the entire hot end. So, you can print one once you get your first printer set up!

Queda fuera de servicio hasta resolverse este problema.

dmarcosm, 2018-06-14:
En una de las sesiones anteriores se colocaron en Geonosis las piezas que imprimió Dédalo. Todo bien exceptuando que el ventilador de capa podía llegar a chocar con los perfiles, así como tener que cambiar la dirección del endstop de X en Marlin. 
Para solventar lo primero, optamos en la siguiente reunión por imprimir unas piezas nuevas (@GioSunrider sabe cuáles son), que ya quedaron colocadas, y que colocaba el ventilador encima del extrusor en lugar de a un lado.

Siotma, 2018-06-16:
Parece ser que las piezas nuevas que menciona @dmarcosm son estas: https://www.thingiverse.com/thing:2828421

Siotma, 2018-06-16:
En la última reunión se reimprimió la pieza que sujeta el ventilador del extrusor, porque había quedado muy frágil con ABS. Al parecer es un problema de diseño porque con PLA también quedaron algunas paredes muy finas. Sin embargo, al montarlo todo sobre la impresora ya quedó más sólido. Para añadir fiabilidad, añadimos pegamento termofusible para que diera consistencia a las piezas una vez colocadas.

El ventilador de capa también está fijado con cola termofusible a la parte superior.

El acabado es muy bueno, y el ventilador de capa consigue enviar bastante aire sin enfriar el extrusor. Se deben realizar pruebas de impresión para confirmar que este problema está por fin resuelto.

Este es el aspecto actual del extrusor de Geonosis:

[image]

dmarcosm, 2018-06-21:
Hoy hemos empezado a afinar un perfil de Slic3rPE para Geonosis. Esta es la pieza que mejor ha podido imprimir tras afinar algunos ajustes: 
[image]
No estoy seguro de si el ventilador de capa está funcionando del todo bien (es posible que se esté saliendo algo de aire por arriba y que a la pieza no le llegue lo suficiente), pero por lo general la pieza ha quedado muy muy fina. Ha impreso a unos 80mm/s

GioSunrider, 2018-11-27:
Necesito actualizacion de est issue, creo que @dmarcosm imprimio ya los soportes de los motores

Siotma, 2018-11-27:
> creo que @dmarcosm imprimio ya los soportes de los motores

los imprimí yo con hija: https://github.com/asrob-uc3m/impresoras-asrob/issues/90

GioSunrider, 2018-12-15:
Por el plan de issues genéricas dejo pendiente lo que hay que hacer. Además que está sigue en #114 y está casi completado. Dejo pendiente a @JorFru encargado del recableado ya que demostró habilidades que el resto no, con un grandísimo resultado.

closed 2018-12-15 by GioSunrider