asrob-uc3m/impresoras-asrob #62: La MEGA de Blacky ha muerto

Siotma, 2018-04-26:
Hay que comprar una nueva Arduino MEGA para Blacky, pues la que tenía ha muerto.

Se puede reutilizar https://github.com/asrob-uc3m/impresoras-asrob/issues/48#issuecomment-372278058 :

> Voy con las placas que he mirado. 
> [OPCION 1](https://www.amazon.es/Tarjeta-Microcontrolador-ATmega2560-ATmega16U2-Compatible/dp/B06Y3ZHPWC/ref=sr_1_3?ie=UTF8&qid=1520853461&sr=8-3&keywords=Arduino+mega&dpID=51w0PuG6wHL&preST=_SY300_QL70_&dpSrc=srch)
> [OPCION 2](https://www.amazon.es/XCSOURCE®-Atmega2560-16AU-Tarjeta-Versión-TE629/dp/B01N51Y3EJ/ref=sr_1_11?ie=UTF8&qid=1520853461&sr=8-11&keywords=Arduino+mega)
> [OPCION 3](https://www.amazon.es/SODIAL-desarrollo-ATMega16U2-Mega2560-Arduino/dp/B00PXOGG74/ref=sr_1_14?ie=UTF8&qid=1520853461&sr=8-14&keywords=Arduino+mega)
> Esto para las placas Mega Imitación, deberíamos valorar tiempos, por eso lo pongo por Amazon.

Adjunto foto:

[image]

Siotma, 2018-04-27:
Alguno tiene Amazon Prime para conseguir la OPCION 1 con envío gratis?

David-Estevez, 2018-04-27:
Independientemente de que se compre una placa nueva, crees que merece la pena reparar la placa quemada?

Siotma, 2018-04-27:
Podría merecer la pena, pero no para ponerla en una impresora, sino más bien para tenerla para prototipar...

David-Estevez, 2018-04-27:
Sí, mi idea era más bien tenerla "de repuesto" por si alguna vez hace falta echarle mano para lo que sea.

dmarcosm, 2018-04-27:
Quizá sería lógico comprobar primero con otra RAMPS (la de Hija, mismamente), que ningún otro componente se ha visto afectado, así si hay que comprar más cosas las compramos de una sola vez, no?

GioSunrider, 2018-04-27:
>Alguno tiene Amazon Prime para conseguir la OPCION 1 con envío gratis?

Yo, y podria comprarla sin problemas. Pero la idea de @dmarcosm me parece muy adecuada.

Siotma, 2018-04-28:
Además, yo creo que la placa se quemó por culpa de un contacto indirecto de la cadena de anclaje con la electrónica. Tiene pinta de que la parte quemada actuó de fusible evitando daños mayores. De todas formas, cuanto antes pidamos la placa, antes podremos hacer pruebas y volver a tener a Blacky operativa.

> Yo, y podria comprarla sin problemas.

Recuerda solicitar factura acorde con la normativa de subvenciones, disponible [aquí](https://www.uc3m.es/ss/Satellite/Cultura/es/TextoDosColumnas/1371240285442/) (ver artículo 12).

GioSunrider, 2018-04-29:
Si sale mas barato adelante, yo en ebay jamas he comprado nada asi que no tengo opiniones.

Siotma, 2018-05-06:
Aceptada la compra entonces. Compraré dos unidades para tener una de repuesto y prototipado.

Siotma, 2018-05-11:
Realicé el pedido y llegó el miércoles. El jueves lo llevé a la nave.

@GioSunrider estuvo realizando el cambio de placa y nos puede contar más detalles...

GioSunrider, 2018-05-15:
Hay pocos detalles que añadir, lo primero se subió el Marlin a la placa sin ningún problema y se conecto a la ramps con comprobación de las conexiones. Una vez instalado todo, se procedio a la conexion de la fuente de alimentacion. Se encendio correctamente y al ir a hace Home All, se apago de repente. En ese punto desconecte la fuente de alimentación y efectivamente olia a quemado y el regulador de la nueva placa habia muerto. @Siotma intento sustituirlo por el de la anterior pero no coincidian asi que se queda como tarea pendiente.

Siotma, 2018-05-19:
En la última reunión, realicé un proceso de diagnóstico de la placa que se quemó:

En la reunión anterior se quemó el regulador de la MEGA, pero comprobé si seguía funcionando por USB: afirmativo, la placa está en buen estado, tan solo le falta el regulador para funcionar alimentada por VIN, por tanto no sirve para las impresoras. El regulador que llevaba es un AMS1117-5.0 formato SOT223. Al desoldar el que tenía, se rompió el pad 2 en la placa, pero está disponible a través del tab, por lo que yo creo que se podría poner uno nuevo. No obstante, no sé si merece la pena ponerle un regulador o dejarla para prototipar por USB. Digo esto, teniendo en cuenta que hay otras dos nuevecitas que compramos (https://github.com/asrob-uc3m/impresoras-asrob/issues/62#issuecomment-386871106).

Actualmente la placa está en el armario, dentro de una bolsa con una nota que explica lo anterior.

Foto: 

[image]

Siotma, 2018-05-19:
Aprovechando que tenía una MEGA funcional, pero sin regulador, pensé que era una oportunidad perfecta para diagnosticar el fallo de Blacky que había acabado con el regulador: En una placa sin regulador, no se puede quemar el regulador, y en caso de sobre consumo, al estar conectada por USB, la protección del puerto apagaría todo el sistema. Así lo hice.

Conecté la placa a la RAMPS de Blacky y fui conectando cables poco a poco, comprobando que funcionaba cada uno:
1. Conecté el extrusor y probé a calentar: funcionaba de forma intermitente. Comprobé que la clema de la RAMPS estaba en mal estado y solo hacía contacto si apretabas con el dedo. Sustituí la clema por una nueva y el problema se solucionó. El extrusor calentaba bien.
2. Conecté la cama y probé a calentar. Todo correcto.
3. Cogí un pololu y fui probando todos los zócalos con él, y con el motor del extrusor: los zócalos de X, Y, Z, E0, funcionan correctamente.
4. Cogí todos los pololus y los fui probando en el zócalo del eje X con el motor del extrusor: todos los pololus están en buen estado.
5. Coloqué cada pololu en su posición y fui probando los ejes, finales de carrera incluidos. Todo correcto. Tan solo detecté que un cable del motor del eje Y estaba despeluchado a la altura del conector y corría riesgo de hacer cortocircuito con los pines cercanos. Cambié el crimp y lo dejé saneado.
6. Conecté la pantalla y probé a controlar la impresora desde ella: funcionó correctamente.

A estas alturas la placa tenía todas las conexiones y funcionaba correctamente. El puerto USB no parecía estar sufriendo. Así que me la jugué, flasheé otra MEGA nuevecita y se la puse, conectando todo poco a poco igual que lo he descrito antes.

El resultado fue satisfactorio y di a Blacky por arreglada, coloqué la placa en su sitio y la pusimos a imprimir:

[image]
[image]

closed 2018-05-19 by Siotma