asrob-uc3m/impresoras-asrob #63: Cable management en Dédalo

Siotma, 2018-04-26:
Los cables de Dédalo son un peligro para todos y para sí misma.

Ahora que le hemos puesto una placa nuevecita, molaría dejar la impresora en condiciones para que dure funcionando mucho tiempo, para ello propongo, al menos:

- [ ] Aumentar la longitud del cable del ventilador de capa, y conducirlo paralelo a los demás cables que van al extrusor.
- [ ] Revisar que ningún cable estorba el movimiento de ningún eje.
- [ ] Imprimir un soporte donde atornillar la OVM20.

- [ ] EXTRA: quitar todas las bridas y retirar el cableado. 
- [ ] EXTRA: Separar todos los cables e identificarlos. 
- [ ] EXTRA: Conducirlos con nuevas bridas por las varillas de manera eficiente.

JorFru, 2018-04-26:
Claramente soy indispensable, me asigno el trabajo de cables.

Siotma, 2018-04-27:
En efecto, los cables que manejan mucha corriente sería importante valorar el cambiarlos por los de silicona que tenemos. 

Respecto a los cables de los motores, termistores y finales de carrera, la sensación que me dio ayer es que son suficientemente largos (algunos incluso sobra un buen trozo), pero que llevan tanto tiempo en la misma posicion que están viciados y acartonados. Yo propongo soltarlos completamente y replantear su trazado. Sobre todo hay que evitar que se enreden y que estorben el movimiento de los ejes.

David-Estevez, 2018-04-27:
Creo que me voy a tener que presentar al título de "Rey de las ñapas"... :see_no_evil: :see_no_evil: :see_no_evil:

Siotma, 2018-05-06:
En la última reunión, @dmarcosm y yo hicimos los siguientes progresos:

- Alargamos el cable del ventilador de capa con los cables existentes en el armario y le pusimos un conector para conectarlo directamente a lo pines de la OVM20Lite.
- Analizamos el cableado de la impresora, y detectamos que había muchos cables con una longitud mayor de la necesaria. Estos cables tenían varias vueltas innecesarias para quedar pegados a las varillas.
- Retiramos la mayor parte de las bridas para liberar los cables y replantear su trazado.
- Recondujimos los cables de:
  - El extrusor, el ventilador de capa y el termistor.
  - El final de carrera del eje Y.
  - El final de carrera del eje Z.

Que @dmarcosm añada si se me ha olvidado algo.

Siotma, 2018-05-06:
Es urgente disponer de un soporte para dejar la placa instalada. Tal vez nos valga el que tenía la antigua placa, aunque sea de forma temporal. @David-Estevez , podrías traerlo?

David-Estevez, 2018-05-06:
Sin problema! :+1:

David-Estevez, 2018-05-07:
Ya los tenéis, están puestos en Dédalo

Siotma, 2018-05-13:
En la reunión del último jueves continuamos revisando la longitud de los cables y fijamos la placa a la impresora. 

Además, colocamos una cadeneta para conducir los cables de potencia y el termistor de la cama.

Cada vez queda menos para volver a tenerla funcionando.

Siotma, 2018-05-19:
@dmarcosm tiene nueva información sobre esto :wink:

dmarcosm, 2018-05-22:
En la última reunión ya quedaron todos los cables con la longitud deseada y medianamanete colocados (hasta que no tengamos una caja para la placa no terminaría de colocarlos).
Estuvo imprimiendo a la perfección (dentro de lo que cabe) unas piezas para el ventilador de capa de Geonosis :)

CinnabarCinnamon, 2019-02-14:
Se ha comenzado el recableado de Dédalo, se ha ajustado (cortar y soldar) las longitudes de los cables de Y y X.

CinnabarCinnamon, 2019-02-14:
El cableado de Z y Extrusor están a medio camino, Dédalo que inutilizable hasta que se acabe.

GioSunrider, 2019-02-14:
Se ha colocado ademas el LCD, como se ve en #129

CinnabarCinnamon, 2019-02-21:
Se da por finalizada la colocación de cables. Cerrando issue.
[image]

closed 2019-02-21 by CinnabarCinnamon

GioSunrider, 2019-02-21:
Se ha hecho una prueba de impresion durante la charla de @JorFru para el GUL.

[image]