asrob-uc3m/impresoras-asrob #64: Comprar regletas para la granja de impresoras

Siotma, 2018-04-29:
Hay que comprar regletas que lleven alimentación hasta la granja de impresoras y sistemas auxiliares.

De momento se ha detectado la necesidad de dos regletas, y se ha medido su longitud:

- [ ] Regleta de 5 metros con mínimo 8 tomas de corriente, e interruptor. (Para las impresoras, el router y las raspberrys).
- [ ] Regleta de 3 metros con mínimo 5 tomas de corriente e interruptor. (Para poder poner hasta 2 PCs, 2 pantallas y la Arcade).

Las compraré esta semana en Bricomart / Leroy Merlin o similar para poder instalarlas en la próxima reuníon.

Siotma, 2018-04-30:
Hoy he comprado dos regletas:

- 5 metros y 6 tomas con interruptor (dado que el router va a estar conectado permanentemente ya no hacen falta tantas tomas, en un futuro se pueden colgar nuevas regletas de esta.
- 3 metros y 6 tomas con interruptor.

Total 13,98€

Siotma, 2018-05-04:
En la reunión de ayer, se instalaron las regletas en su lugar:

[image]

[image]

closed 2018-05-04 by Siotma

David-Estevez, 2018-05-04:
Puede que sea conveniente hacerles algún tipo de ASROB-marca para que se sepa que son nuestras (y que, concretamente, son de la granja).