asrob-uc3m/impresoras-asrob #67: Comprar un MOSFET para Geonosis.

Siotma, 2018-04-29:
@JorFru dijo en #60 que sería buena idea ponerle un MOSFET a Geonosis cuanto antes.

He encontrado [este](https://www.ebay.es/itm/MKS-MOSFET-Extrusor-Cama-Caliente-Impresora-3D-Heating-Control-Bed-Extruder-I093-/202154694034) de un proveedor de confianza de eBay. Sin embargo, los que tenéis una Anet estáis más puestos que yo en el tema...

Tareas:
- [ ] Discusión y selección de un MOSFET adecuado.
- [ ] Realizar el pedido del MOSFET.
- [ ] Recepcionar el MOSFET.
- [ ] Instalar el MOSFET en Geonosis.

dmarcosm, 2018-04-30:
Ese mismo que propones es el que he comprado yo para mi Anet @Siotma. Diría que es el indicado, aunque creo que @JorFru puede aportar más en este caso.

JorFru, 2018-04-30:
Todos cumplen con lo que necesitamos

GioSunrider, 2018-04-30:
En mi caso el que esta es [este](https://www.amazon.es/gp/product/B0714JXQT7/ref=oh_aui_detailpage_o03_s00?ie=UTF8&psc=1). Sin ningun problema en envio y sin ninguna queja de el.

Siotma, 2018-05-06:
Gracias por vuestro asesoramiento. Damos por bueno entonces el modelo seleccionado.

Siotma, 2018-05-11:
Realicé el pedido y llegó el miércoles. El jueves lo llevé a la nave y @GioSunrider nos tiene qué contar qué tal se le dio montarlo.

GioSunrider, 2018-05-15:
Instale el MOSFET sin ningún problema. siguiendo el esquema de conexiones de la siguiente foto.

[image]

Para ello tuve que añadir la tercera conexión de la fuente de alimentación, ademas de meter también los cables del ventilador que estaban un poco mal sacados, y fijarlos al marco de Geonosis. Adjunto foto del resultado.

[image]

Ademas se probo a ver si calentaba la cama y sin ningun problema. Asi que cierro issue.

closed 2018-05-15 by GioSunrider