asrob-uc3m/impresoras-asrob #68: Comprar una fuente de alimentación decente para Hija

Siotma, 2018-04-29:
Todos nos acordamos de aquella mala racha de fuentes de alimentación #28 #33 #35 etc..

La fuente de alimentación actual de Hija Resurrection no es de las más prometedoras de las que tenemos. Además, dada su forma, no podemos fijarla a la estructura de la impresora. Según tengo entendido, fuentes de alimentación de este estilo son fiables y además se pueden fijar en impresoras P3steel como Hija:

[image]

He encontrado varios proveedores:
- eBay [uno](https://www.ebay.es/itm/Fuente-de-alimentacion-12V-30A-360W-Impresora-3D-Switching-Power-Supply-I0015-/201636784333) y [dos](https://www.ebay.es/itm/12V-30A-360W-Impresora-3D-Fuente-conmutada-ALIMENTADOR-LED-TRANSFORMADOR-I0015-/201998587004)
- orballoprinting [uno](https://orballoprinting.com/gl/electronica/77-fuente-de-alimentacion-300w-25a.html)

Tareas:
- [ ] Seleccionar el modelo de fuente compatible con el requisito de que debe ser fácil de instalar en la impresora.
- [ ] Realizar el pedido.
- [ ] Recibir el pedido.
- [ ] Instalar la fuente de alimentación en Hija Resurrection

Siotma, 2018-05-06:
A mi me mola la de HTA3D

GioSunrider, 2018-05-06:
A mi me parece también la mejor opción.

Siotma, 2018-05-25:
En la reunión de ayer se comenzó el montaje de la fuente de alimentación de Hija Resurrection.

Fue necesario modificar el cableado que hizo @GioSunrider porque la luz del interruptor se encendía constantemente. Con el nuevo cableado solamente se enciende cuando está en la posoción ON.

También ajustamos la posición de los agujeros para fijar las piezas entre sí con tornillos.

Calibré la tensión de salida a 12,20 V.

Pusimos el cable de alimentación de la RAMPS. Falta otro para la cama caliente y otro para los ventiladores y otros sistemas futuros.

Este es el estado actual:

[image]

Siotma, 2018-06-03:
En la última reunión se finalizó el montaje de la fuente de alimentación de Hija Resurrection.

Se colocaron 3 parejas de cables:
1. AWG16 rojo y negro para la RAMPS (alimentación de la electrónica, potencia del extrusor, motores).
2. AWG16 rojo y negro para la cama caliente (potencia de la cama a través del relé).
3. AWG18 (más finos) para los ventiladores.

La fuente se montó ancló a la impresora con 2 tornillos M4.

[image]

Los cables que salen de la fuente son demasiado largos, pero esto es adrede, la idea es que tras el cable management de la impresora, veamos cuál es la longitud necesaria y en su caso, los cortemos a medida.

[image]

Los ventiladores se conectorizaron utilizando conectores JST disponibles en el armario, y las bananas que usaban antes, se han devuelto a una de las cajas componentes electrónicos.

[image]

Se probó el correcto funcionamiento de la fuente de alimentación, probando a calentar la impresora y mover todos los motores.

La fuente que tenía Hija Resurrection ahora está en el armario, y yo propongo que se la pogamos a Blacky...

Issue cerrada.

closed 2018-06-03 by Siotma

Siotma, 2018-10-14:
Piezas actualizadas via 0363e391158485261eb205cfc239e86b738a5702