asrob-uc3m/impresoras-asrob #74: Comprar estación de soldadura

dmarcosm, 2018-05-08:
He visto esta estación de soldadura, con pistola de aire caliente incluída, y creo que nos podría venir muy bien para la mesa de soldadura: https://www.impextrom.com/es/estacion-de-soldadura-digital-doble-operaciones-smd-baku-601d-p999982500

¿Os parece buena idea comprarla? Se aceptan sugerencias de los masters de la soldadura :D

JorFru, 2018-05-09:
Yo la tengo. la he dado mucho trote y aguanta. Tiene potencia de sobra y el aire es cómodo de usar.
Las puntas que trae son KK, pero sirven para tirar un año o dos.

David-Estevez, 2018-05-09:
Yo también la tengo y le doy mis bendiciones.

GioSunrider, 2018-05-10:
Yo doy tambien doy las bendiciones, con el feedback de @JorFru y @David-Estevez, me parece la idea cojonuda.

David-Estevez, 2018-05-10:
Si os parece guay la idea, el siguiente paso sería buscar financiación para la estación.

GioSunrider, 2018-10-08:
Traslado a [Actas #138](https://github.com/asrob-uc3m/actas/issues/138)

closed 2018-10-08 by GioSunrider