asrob-uc3m/impresoras-asrob #81: Calidad de las impresiones de Dedalo

GioSunrider, 2018-07-02:
Ultimamente Dedalo, cada vez que imprime tiene perdida de pasos del motor y su calidad no es la de esperar. Pongo foto del ultimo resultado. Hay que investigar mas este problema.

[image]

dmarcosm, 2018-10-05:
Al cambiar la placa de Dédalo a la OVM20 Lite, hemos ganado el soporte para ajustar los drivers de los motores con los potenciómetros que esta tiene. Sería un buen primer paso ajustarlos para ver si se corrigen esas pérdidas de pasos :+1:

JorFru, 2018-10-08:
Hemos subido un poquito la corriente de los drivers. 
Hemos seguido esta guía. http://www.dima3d.com/motores-paso-a-paso-en-impresion-3d-iii-calibracion-de-corriente/
La OVM en teoría es: I = V x 2
Antes de tocar hemos medido 0.250V en cada driver. Eso son 0.5A

Lo hemos dejado en 0.350V en cada driver (excepto en Z que lo hemos dejado a 0.250V). Lo cual son 0.700A de corriente.
Hemos notado que los drivers se calentaban un poco más, pero no más de 60ºC
Con ventilación de electrónica no se nota ninguna temperatura anómala.

dmarcosm, 2018-10-08:
Al subirlos ha mejorado la impresión?

GioSunrider, 2018-12-15:
Dédalo en principio está operativa. Procedo a cerrar la issue.

closed 2018-12-15 by GioSunrider