asrob-uc3m/impresoras-asrob #82: It's a trap

JorFru, 2018-09-13:


JorFru, 2018-09-16:
[image]

JorFru, 2018-09-17:
[image]
Spicy caldito from la abuela

JorFru, 2018-09-17:
La ANET llegó justo en el momento oportuno. En La Nave de ASROB se congregaron nueve nuevas personas en la primera reunión de PrinterOne del curso
[image]
[image]
Vídeos del montaje:
https://www.youtube.com/watch?v=dK14sc66EE8
https://www.youtube.com/watch?v=Sgs6Ly7aKj8

dmarcosm, 2018-09-19:
Añadida carpeta de la Anet al repositorio de impresoras! https://github.com/asrob-uc3m/impresoras-asrob/tree/master/Anet

David-Estevez, 2018-09-19:
Hay que darla de alta en la web/bot?

dmarcosm, 2018-09-19:
> Hay que darla de alta en la web/bot?

Yo no lo haría hasta que tengamos nombre definitivo. Así evitamos luego tener que entrar a cambiarlo otra vez :D

GioSunrider, 2018-10-03:
Foto de la Anet acabada.

[image]

dmarcosm, 2018-10-16:
> Hay que darla de alta en la web/bot?

Ahora sí hay que darle de alta! Spicyyy!!

JorFru, 2018-10-16:
Hoy se ha hecho el cable management.
Asigno a la gente que me acuerdo que ha participado. Quien falte, que se apunte.

Foto del logro desbloqueado:
[image]

GioSunrider, 2018-10-16:
> > Hay que darla de alta en la web/bot?
> 
> Ahora sí hay que darle de alta! Spicyyy!!

Ya esta la issue puesta, keep calm, hoy le hemos hecho las fotos. Mañana las subire y se le podra dar de alta.

David-Estevez, 2018-10-21:
Ya está dada de alta en la web