asrob-uc3m/impresoras-asrob #83: Cambio Extrusor Blacky

GioSunrider, 2018-09-13:
Al igual que en hija se propone un cambio completo del extrusor. @Siotma tiene un modelo que iria perfecto para blacky, por lo que le invoco para que como en hija nos diga las piezas y el modelo.

Siotma, 2018-09-14:
La idea de este extrusor es que lleva el filamento a través de un mismo tubo de teflón desde la zona de empuje directamente hasta la zona caliente, por lo que no hay interfaces donde pueda quedar atascado. Desatascar este extrusor es tan fácil como cortar el trozo de tubo donde esté el atasco, en el peor de los casos.
En mi experiencia esto no es necesario ya que el plástico desliza por el interior del tubo sin problema aunque esté frío: he llegado a quitarle el filamento a mi impresora sin calentar y solamente hay que hacer una pequeña tracción para conseguirlo.

El concepto es el siguiente:

[image]

Siotma, 2018-09-19:
Sí, mi intención es que ambas impresoras utilicen el mismo sistema. 
**Pero**, una vez instaladas las piezas de #84 en Hija, se debería revisar qué modificaciones o mejoras se quieren añadir antes de imprimir un nuevo kit para Blacky.

Siotma, 2018-10-16:
Lo pongo para considerarlo: Blacky es del departamento y ponerle un nuevo extrusor puede ser un cambio bastante significativo... Tal vez habría que consultarlo y/o solicitar la financiación del extrusor por parte del Departamento para que no corran el riesgo de ir perdiendo porcentaje de posesión sobre la impresora :laughing:

GioSunrider, 2018-10-17:
Me parece perfecto, @David-Estevez, ¿podrias ponerte tu en contacto par ver la solucion del tema?

David-Estevez, 2018-10-29:
Lo he consultado con el jefe de departamento y me ha dado el ok tanto al cambio como a la financiación por parte del departamento. 

Pasadme los enlaces de lo que queréis comprar y hacemos el pedido.

Siotma, 2018-10-31:
La modalidad que yo propongo sería la de trueque: el extrusor ya lo hemos comprado con la subvención, pero nos hacen falta otras piezas para otras impresoras. En breve postearé una lista con enlaces.

Siotma, 2018-11-07:
El extrusor costó 9.24€ (ver [aquí](https://github.com/asrob-uc3m/impresoras-asrob/issues/85#issuecomment-425716268)). Sin embargo, un extrusor equivalente vale más de 20€ en España (ver [aquí](https://fervi3d.com/extrusion/116-extrusor-e3dv6-completo-clon.html)).

Propongo como trueque la compra de 7 unidades de rodamiento **Igus RJ4JP 01-08 - Equivalente a Lm8uu** por un valor de 1,30€/unidad, total 9,10€ más envío.

Enlace al artículo: https://fervi3d.com/mecanica/380-igus-rj4jp-01-08-equivalente-a-lm8uu.html

El proveedor propuesto es solamente orientativo. Si se encuentra el mismo producto en otro proveedor más conveniente no habría problema.

GioSunrider, 2018-12-15:
Sigue en #45

GioSunrider, 2018-12-15:
Reabrir para que cuando pueda @David-Estevez nos confirme la decisión del departamento.

GioSunrider, 2019-01-13:
Bloquea a #52 cuando @David-Estevez nos confirme dicha decision.

David-Estevez, 2019-02-07:
Los rodamientos han sido pedidos. Os aviso cuando lleguen.

David-Estevez, 2019-02-07:
Seguimos en #52

closed 2019-02-07 by David-Estevez