asrob-uc3m/impresoras-asrob #84: Nuevo extrusor para Hija

Siotma, 2018-09-14:
En la próxima reunión llevaré un Hotend para montar en Hija.

Sería ideal que además se llevaran impresas las siguientes piezas:

[1x Carro del eje X y 1x prensa de la correa](https://www.thingiverse.com/thing:701087)

[1x Wade_L3K_custom_flange_1.75_mirrored.stl.zip](https://github.com/asrob-uc3m/impresoras-asrob/files/2384180/Wade_L3K_custom_flange_1.75_mirrored.stl.zip)

[1x Wade_L3K_custom_body_1.75_centered.stl.zip](https://github.com/asrob-uc3m/impresoras-asrob/files/2384181/Wade_L3K_custom_body_1.75_centered.stl.zip)

En realidad las dos últimas son son un remix que hice de https://www.thingiverse.com/thing:512338, pero diseñado para poder meter el tubo de teflón hasta el Barrel

Que empiece el reto!

Siotma, 2018-09-14:
La idea es que los que estén interesados en imprimir cualquiera de las piezas, se apunten con un comentario para evitar duplicaciones...

dmarcosm, 2018-09-18:
Este carriage es al que hay que meterle una tuerca a mitad de impresión? Si es así, que tipo de tuerca exactamente? 
Yo puedo imprimir un set entero, pero aviso de que será morado/rosa de BQ.

dmarcosm, 2018-09-18:
Antes de imprimir creo que mejor estar seguros. Hemos estado mirando los STLs y no nos cuadra que sea para un Bowden, básicamente porque la pieza que sujeta el motor y la que sujeta el hotend es la misma, o sea que sería directo.
Entonces, nos planteamos que vamos a seguir con un extrusor directo en vez de un bowden o qué hacemos @Siotma  
Por otra parte, faltan piezas como los engranajes y la pieza que sujeta el rodamiento, o reutilizamos las que ya tiene la impresora?

Siotma, 2018-09-18:
La tuerca es una M4, pensaba que lo ponía en la página de la pieza. Cuando lo imprimí yo, no me cabía M4 y tuve que poner M3, que también va bien pero baila un poco.

Los STLs tienen la ventaja de servir **tanto para extrusión directa como para bowden**. En un primer momento lo pondríamos como directo y una vez la impresora esté funcional se puede imprimir un par de piezas para convertirlo en bowden.

Los engranajes yo diría que se pueden reutilizar los del mismo extrusor, porque ya lo hicimos con Blacky.

Lo que sí se me olvidó fueron las piezas del idler.

Voy a actualizar el primer post con nuevas piezas.

Siotma, 2018-09-18:
Tal y como se puede ver en [esta issue](https://github.com/asrob-uc3m/impresoras-asrob/issues/45#issuecomment-368879177), no hace falta cambiar los engranajes.

Siotma, 2018-09-18:
Actualizado primer post con más información.

Siotma, 2018-09-18:
Aún más piezas disponibles ! 

Recomiendo no hacer más de un kit hasta que lo hayamos probado y visto los problemas...

dmarcosm, 2018-09-19:
He añadido los botoncitos de checklist para ir marcando las que hayamos impreso para este set. Para evitar esto basicamente: 

> La idea es que los que estén interesados en imprimir cualquiera de las piezas, se apunten con un comentario para evitar duplicaciones...

dmarcosm, 2018-09-19:
Like por los soportes ya incluídos en el STL de la pieza más grande. Acabo de descubrir que no eran parte de la pieza :D

dmarcosm, 2018-09-20:
[image]

Siotma, 2018-09-28:
Las piezas se montaron en la reunión del pasado lunes, junto con un extrusor que aporté yo mientras se gestiona la compra de los nuevos en #85 

El montaje mecánico quedó resuelto, tan solo falta apretar bien el hoten en caliente para que no se desenrosque, y hacer el cableado. Este es el estado actual:
[image]

Entre las ventajas que presenta este extrusor está el hecho de que no hay que andar enroscando y desenroscando tornillos para meter y sacar el filamento, sino que basta con retirar la lengüeta con los dedos para liberarlo:

[image]
[image]

El tubo de teflón va desde el hobbed bolt hasta el heatblock por lo que el filamento nunca cambia de interfaz mientras está frío, lo que previene los atascos:

[image]

dmarcosm, 2018-09-28:
> En lo que respecta a esta issue, @dmarcosm debería tener el honor de cerrarla

No! Falta la tobera del ventilador!!

MrTommyGuns, 2018-10-06:
En la reunión no llegamos a apretar el hotend (no sé se si hizo fuera de ella) y los cables falta ponerlos bien con bridas (no dio tiempo)

JorFru, 2018-10-06:
Gracias Tommy, queda pendiente entonces, editado.
Cuando termiéis esta issue, acuérdate de marcar los checkboxes de arriba y cuando esté todo completo (ventilador de capa instalado) se podrá cerrar la issue.

Siotma, 2018-10-13:
En la última reunión (8 de octubre), continué los trabajos en Hija:
- Deshice el apaño del cableado, ya que podía quedar pellizcado entre la estructura. Se acordó que el cableado del extrusor se debe guiar con una cadeneta: #66 #99 
- Soldé el ventilador blower de 40x40x20 para el ventilador de capa.
- Añadí una vuelta de cinta aislante en la zona donde se ancla el hotend para evitar que se mueva, ahora queda bien fijo.
- Apreté a conciencia el hotend, en frío y luego en caliente.
- Realicé una prueba de cubo de impresión, satisfactoria:

[image]

- También comprobé que el nuevo extrusor admite retracciones grandes sin problema de atascos: llegué a probar más de 20mm de retracción.
- Probé con éxito a insertar el filamente utilizando el motor del extrusor.
- Probé con éxito a extraer el filamento utilizando el motor del extrusor.
- Probé con éxito a extraer el filamento en frío utilizando el motor del extrusor.

Siotma, 2018-10-13:
Tras ello, puse a Hija a imprimir su propio ventilador de capa, que quedó estupendamente:

[image]

Siotma, 2018-10-14:
Piezas actualizadas via 0363e391158485261eb205cfc239e86b738a5702

Siotma, 2018-10-16:
Hoy se ha completado el montaje del ventilador de capa. Se han realizado [dos pruebas de impresión](https://www.thingiverse.com/thing:922740) con Hija Resurrection que han dado como resultado una nueva caja para RaspberryPi:

[image]

La renovación del extrusor de Hija ha sido todo un éxito y se puede dar la issue por cerrada. @dmarcosm procede.

closed 2018-10-16 by dmarcosm