asrob-uc3m/impresoras-asrob #85: Comprar nuevos extrusores

Siotma, 2018-09-14:
La idea es comprar extrusores de estos si la experiencia de la próxima semana es buena.

https://es.aliexpress.com/item/Best-price-J-head-Hotend-with-cooling-Fan-for-1-75-3-0mm-12v-V6-bowden/32810935651.html?spm=a219c.10010108.1000016.1.2ac25f11dnjyT2&isOrigTitle=true

Están pensados tanto para extrusión directa como para Bowden. La idea de este extrusor es que lleva el filamento a través de un mismo tubo de teflón desde la zona de empuje directamente hasta la zona caliente, por lo que no hay interfaces donde pueda quedar atascado. Desatascar este extrusor es tan fácil como cortar el trozo de tubo donde esté el atasco, en el peor de los casos.
En mi experiencia esto no es necesario ya que el plástico desliza por el interior del tubo sin problema aunque esté frío: he llegado a quitarle el filamento a mi impresora sin calentar y solamente hay que hacer una pequeña tracción para conseguirlo.

El concepto es el siguiente:

[image]

Además, yo añadiría al pedido:

- [Tubo de teflón extra](https://es.aliexpress.com/item/1M-PTFE-tube-Teflon-Pipe-for-1-75mm-or-3mm-Filament-of-3D-Printer-Parts/32818240272.html?spm=a2g0s.13010208.99999999.262.1eb83c00zgzF8T).
- [Barrels extra](https://es.aliexpress.com/store/product/5-pcs-bowden-wade-V6-hotend-Heat-Break-Throat-for-1-75-3-00-mm-Filament/1494743_32810212860.html?spm=a219c.12010612.8148356.3.6734383cqJGwSO).
- [Heatblocks extra](https://es.aliexpress.com/store/product/3D-printer-heating-block-V6-heating-block-fit-j-head-hotend-kossel-and-prusa-i3/1494743_32812503617.html?spm=a219c.12010612.8148356.22.37e93d15eFdgfE).
- [Alguna pieza de silicona antiadherente para probar](https://es.aliexpress.com/store/product/1-PC-New-E3D-V6-Silicone-Sock-heating-block-set-1-75-3-0mm-Heated-Block/1494743_32865931652.html?spm=a219c.12010612.8148356.1.37e93d15eFdgfE)

Estimo un precio de unos 40-45€ por todo. La idea es aumentar la robustez y la fiabilidad de las máquinas, por lo que los gastos se podrían cargar a la subvención de Granja de Impresoras 3D.

Acciones a realizar:

- [ ] Discutir la cantidad de extrusores a comprar y las piezas extra.
- [ ] Realizar el pedido.
- [ ] Recibir el pedido.

Siotma, 2018-09-26:
En la última reunión llevé un extrusor como el que comento que tenía por casa. Lo instalé y quedó bastante bien en Hija, además los asistentes mostraron interés en comprarlo para las demás impresoras.

Propongo una lista de la compra que sería:

| Item | Precio | Cantidad | Subtotal |
|:-:|:-:|:-:|:-:|
| Extrusor | 9.14€ | 3 | 27.42€ |
| Racores | 0.43€ | 3 | 1.29€ |
| 1m de teflón | 0.75€ | 2 | 1.5€ |
| Barrels | 0.50€ | 5 | 2.5€ |
| Heatblocks + funda | 2.04€ | 5 | 10.20€ |
| Heatblocks vulcano | 0.81€ | 2 | 1.62€ |
| Envío | 2.37€ | 1 | 2.37€ |
| **TOTAL** |   |   | **44.38€** |

Con esto, yo creo que habría material para extrusores por un largo tiempo.

Opiniones!

GioSunrider, 2018-09-27:
ADELANTE!

Siotma, 2018-09-28:
Alguna otra opinión? Podemos añadir [algún](https://es.aliexpress.com/store/product/1-pcs-4020-Blower-Cooling-Fan-40x40x20mm-DC-12V-3D-printer-parts-J-head-hotend-Reprap/1494743_32809789302.html?spm=a219c.12010615/itm2home-1.8148356.17.5b9f25b2ohWc5f) [ventilador](https://es.aliexpress.com/item/5015-12-V-y-24-V-refrigeraci-n-Turbo-ventilador-sin-escobillas-3D-impresora-2Pin-para/32874940278.html?spm=a219c.12010615%2Fitm2home-1.8148356.9.5b9f25b2EFYEQn) yo creo ...

dmarcosm, 2018-09-28:
Aparte del ventilador (para Geonosis), quizá estaría bien comprar algún kit de nozzles y agujas?

Siotma, 2018-09-30:
> kit de nozzles y agujas

Voy a ver si encuentro algo en este vendedor...

Siotma, 2018-09-30:
Pedido!

He añadido los ventiladores, los [nozzles de 0.4 y de 0.5](https://es.aliexpress.com/item/5-pcs-3D-printer-V5-V6-nozzle-for-1-75mm-filament/32811224171.html?spm=a2g0s.9042311.0.0.3d1363c0rzcCp7) y también [agujas](https://es.aliexpress.com/item/Best-peice-10-pcs-Nozzle-Cleaning-0-2mm-1-2mm-Drill-Bits-for-3D-Printer-Nozzle/32812095832.html?spm=a2g0s.9042311.0.0.3d1363c0rzcCp7).

| Item | Precio | Cantidad | Subtotal |
|:-:|:-:|:-:|:-:|
| Extrusor | 9.24€ | 3 | 27.72€ |
| Racores | 0.43€ | 3 | 1.29€ |
| 1m de teflón | 0.75€ | 2 | 1.5€ |
| Barrels | 0.30€ | 5 | 1.5€ |
| Heatblocks + funda | 2.06€ | 5 | 10.30€ |
| Heatblocks vulcano | 0.82€ | 2 | 1.64€ |
| Ventilador 40x40x20 | 1.13€ | 1 | 1.13€ |
| Ventilador 50x50x15 | 1.04€ | 1 | 1.04€ |
| Nozzle 0.4 | 0.12€ | 5 | 0.57€ |
| Nozzle 0.5 | 0.12€ | 5 | 0.57€ |
| Kit 10 agujas 0.2 0.3 0.4 | 4.00€ | 1 | 4.00€ |
| Envío | 3.40€ | 1 | 3.40€ |
| Descuento | 1.75€ | 1 | 1.75€ |
| **TOTAL** |   |   | **52.80€** |

Siotma, 2018-10-11:
Me he dado cuenta de que esta compra tiene más sentido ponerla en la subvención de Modernización. Hay presupuesto suficiente así que presentaré el gasto contra esa subvención.

GioSunrider, 2018-10-14:
Pregunta, de este kit, sobraba un extrusor? es que he caido que una mejora para spicy seria comprarle uno de estos y ponerla bowden.

Siotma, 2018-10-16:
Se han comprado 3 extrusores:
- 1 es para reponerme a mi el de Hija.
- Los otros dos están disponibles. Habría que ver dónde los colocamos. 
  - Sería interesante ponerle uno a Dédalo por Bowden, ya que mejoraría muchísimo su volumen de impresión.
  - El otro hay que pensarlo...

GioSunrider, 2018-10-17:
Por mi parte dejaria a Dedalo en directo por varias razones:
 - La primera es que me gustaria dejar una impresora en directo, pendientes de lo que pase con Blacky en #83 
 - La segunda es que pensando que hay mucho filamento de 3mm el cambio seria bastante a largo plazo. No me gustaria tener en el armario un extrusor pudiendo mejorar las otras.
 - La tercera es que pasaria lo mismo con Blacky. Habría que ver si @David-Estevez le parece adecuado. (En el caso de Spicy pasaria lo mismo con @JorFru).
 - Creo que Spicy mejora una verdadera barbaridad la calidad de impresion colocandole bowden.

dmarcosm, 2018-10-17:
Yo también soy partidario de ponérselo a Dedalo. Primero, tendríamos el mismo diámetro de filamento en todas las impresoras. No hay una necesidad extrema de usar los filamentos de 3mm que tenemos, porque están todos bastante "pasados" digamos. Y segundo por lo que comenta @Siotma del espacio de impresión. Ahora mismo no llega a los 100mm en Z...
El segundo se lo pondría a la Anet pero dejándolo directo.

Siotma, 2018-10-19:
Hoy ha llegado el pedido. Lo llevaré en la próxima reunión.

[image]

Siotma, 2018-10-19:
La discusión sobre en qué impresora se deben instalar los extrusores da para una nueva issue: #107

Esta, que era para la compra, se da por cerrada.

closed 2018-10-19 by Siotma