asrob-uc3m/impresoras-asrob #86: Cristal para Geonosis

GioSunrider, 2018-09-15:
Dado que Geonosis se encuentra actualmente sin cristal, me propongo a gestionar su colocación. Para ello dispongo de un cristal (de nevera) de grandes dimensiones que actualmente se encuentra en mi pueblo. El próximo día que vaya para allá lo traeré y solo haría falta con un cortavidrios, del que también dispongo hacerlo a medida.

IviPartida, 2018-09-15:
Perfecto, el dia que lo hagas si necesitas ayuda cuenta conmigo🤙🏾

GioSunrider, 2018-10-17:
Ayer lleve (Por fin) el cristl para Geonosis, con la ayuda de @MrTommyGuns y @L4g0 cortamos el cristal y se instalo en geonosis.

[image]

closed 2018-10-17 by GioSunrider