asrob-uc3m/impresoras-asrob #87: OVM20 Anet Edition beta from Staticboards

dmarcosm, 2018-09-17:
Le he comentado la posibilidad a Javier, de staticboards, de que Printer One forme parte del equipo de testers de la nueva placa OVM20 Anet Edition que están preparando. Le he comentado que tenemos una Anet A8 y una Tronxy X5S, que esta última utiliza una placa réplica de la de la Anet y que quizá estaría bien testear la placa en otros modelos que no sean la Anet A8. 
La idea le ha gustado y el precio sería el mismo que para los demás testers: 20€. Considerando los precios que tienen estas placas habitualmente, he de decir que es un precio más que aceptable. El total con envío incluído serían 26€. 
Los cambios de esta placa con respecto a la original de Anet (o Tronxy en este caso) son, que recuerde:
- Mosfet de calidad para evitar incendios que quemen un edificio de la uni otra vez
- Drivers TMC2208 para los motores de los ejes X e Y (silenciosos) 
- Electrónica de calidad
- Más que no recuerdo

Estaría bien discutir si aceptamos y ver quién la podría comprar.

JorFru, 2018-09-18:
Este vídeo nos dará el SAV necesario para comprarla.
https://youtu.be/5a57VacYQJY

Siotma, 2018-09-18:
Apoyo la compra. De eso va Printer One, de investigar y probar cosas nuevas.

Podría encargarme de la compra, pero mi disponibilidad es reducida. Si me dejais por aquí las instrucciones mejor que mejor.

dmarcosm, 2018-09-18:
Gracias @Siotma! Imagino que habría que enviarla a la uni. Si me dáis la dirección concreta le digo a Javier que me haga la proforma y te paso el link directamente para que pagues Manu, así no te es demasiado problema!

Siotma, 2018-09-18:
La dirección de facturación la encontrarás aquí: https://www.uc3m.es/ss/Satellite/Cultura/es/TextoDosColumnas/1371240285442/

Artículo 12, B.

La dirección de envío, ponedla a casa de @GioSunrider , o al despacho de @David-Estevez .

GioSunrider, 2018-09-18:
Cuanto tarda la placa? @dmarcosm no me importa recibirla, pero dependiendo igual nos merece la pena que llegue a la uni.

JorFru, 2018-09-18:
Enviadla al casillero de David Estévez, no tiene fallo.

También me apunto habilitar un casillero para ASROB. UC3Music tiene uno y
funciona divino.

GioSunrider, 2018-09-18:
La verdad que el casillero seria muy clave, así tampoco dependemos de David. Y podemos enviar todo lo relacionado allí para que cualquiera lo pueda coger.

Siotma, 2018-09-18:
La verdad es que un casillero sería muy interesante. Se merece una Issue en Actas! :wink:

JorFru, 2018-09-18:
> La verdad es que un casillero sería muy interesante. Se merece una Issue en Actas! wink

Podréis contar con ello, he abierto https://github.com/asrob-uc3m/actas/issues/156
Pero no estará disponible para este pedido. Conted con él para el siguiente.

JorFru, 2018-09-19:
Leed https://github.com/asrob-uc3m/actas/issues/156#issuecomment-422821575
tenemos casillero pero en el Sabatini.

Siotma, 2018-09-19:
Pago realizado!

Siotma, 2018-09-28:
Artículo recibido!

[image]

closed 2018-09-28 by Siotma