asrob-uc3m/impresoras-asrob #88: Propuestas y votación nombre nueva impresora - Anet A8

dmarcosm, 2018-09-19:
Esta issue es para proponer nombres para la nueva impresora y su posterior votación. Dejad los nombres propuestos en un comentario y antes de la próxima reunión votaremos las propuestas que haya para dejar el firmware cambiado de forma definitiva.

JorFru, 2018-09-19:
Yo le daría un nombre que recuerde la habilidad innata que tiene esta máquina por arder espontáneamente.

IviPartida, 2018-09-19:
Easy Fire

dmarcosm, 2018-09-19:
Charmander

David-Estevez, 2018-09-19:
Mustafar

JorFru, 2018-09-19:
A mí me mola el rollo SW
Como es negra, podría llamarse Darth Fire

David-Estevez, 2018-09-19:
Red Hot Chili Printer

GioSunrider, 2018-09-19:
Kaiburr, son los cristales que dan poder a los sables de luz.

L4g0, 2018-09-19:
Spicy 💯

dmarcosm, 2018-09-26:
Voten! Abierto hasta el domingo!
https://goo.gl/forms/FsQwZYA145B3FTd93

GioSunrider, 2018-09-27:
Nuevo formulario, debido a los errores del anterior. Queda abieto hasta el domingo.
https://goo.gl/forms/gM2HX9JpW5oOw5Cd2

GioSunrider, 2018-09-30:
El resultado de la votación ha sido el siguiente:

[image]

Luego el nuevo nombre de la impresora es Spicy. Se ha añadido la etiqueta correspondiente y asigno a @dmarcosm para que realice el cambio de carpeta correspondiente.

closed 2018-09-30 by GioSunrider

dmarcosm, 2018-10-05:
> asigno a @dmarcosm para que realice el cambio de carpeta correspondiente.

Hecho