asrob-uc3m/impresoras-asrob #89: Actualización firmware de Spicy Caldito

JorFru, 2018-09-19:
@GioSunrider cambió el firmware de Spicy Caldito, del firm original de la ANET a un Marlin modificado por él, que no es la última versión.

Los resultados antes y después del cambio de firmware:
[image]
El calicat de la izquierda es con el firm original, el de la derecha es con el Marlin de Gio.
Ambos el mismo GCODE

Presiento que hay que ajustar aceleraciones.

dmarcosm, 2018-09-19:
Creo que esto deberíamos ponerlo en la issue de puesta a punto, para evitar tener 8 issues solo de la Anet :(

JorFru, 2018-09-19:
Organízalo tú entonces
Mi idea era abrir una issue para cada parte de la máquina.

- Firmware/placa
- Extrusor
- mecánica

por ejemplo, como hay con las otras máquinas.

Siotma, 2018-09-19:
Un gran número de issues no son un problema si solamente se requiere una o dos acciones máximo para resolverlas. Las issues estilo "puesta a punto de X" no han demostrado funcionar porque se pierden los avances en las diferentes tareas y los nuevos las ven con miedo de todo el trabajo que hay que hacer. Pienso que el sistema de issues está concebido para un enfoque como el que propongo.

Los veteranos deberíamos intentar abrir issues lo más "atómicas" posibles, aunque sean muchas. De esta manera cada una no duraría abierta más de una o dos semanas desde que se empieza a trabajar hasta que se termina, mientras que ahora tenemos issues a medias, que llevan abiertas meses y nadie se acuerda de por dónde se iba.

dmarcosm, 2018-09-20:
> Organízalo tú entonces
> Mi idea era abrir una issue para cada parte de la máquina.
> 
> * Firmware/placa
> * Extrusor
> * mecánica
> 
> por ejemplo, como hay con las otras máquinas.

No lo decía con ánimo de ofender a nadie. 

> ... Las issues estilo "puesta a punto de X" no han demostrado funcionar ...

Quizá entonces deberíamos replantear los "puesta a punto de X" como proyectos y no como issues, ya que se quedan ahí eternamente sin ser cerrados nunca.

dmarcosm, 2018-09-30:
En la próxima actualización hay que aprovechar a cambiar el nombre por Spicy, tal y como ha salido elegido en #88

JorFru, 2018-10-01:
Actualmente está flasheada con el nombre "Spicy", creo recordar.

En esta issue voy a documentar el tema de ajuste de aceleraciones en Spicy. Ya que desde que se cambió el firmware original, no imprime igual de bien, y posiblemente sea esta razón. Buscaré los ajustes que traía por defecto y los pondré en el Marlin que tiene actualmente. Creo que se puede hacer desde la pantalla, o hace falta flashear.

dmarcosm, 2018-10-01:
Aunque se pueda hacer desde la pantalla, diría que mejor dejarlo cambiado en el firmware (por lo que pueda pasar). Así mismo, aprovecharía para actualizar a Marlin 1.1.9 :D

GioSunrider, 2018-10-01:
Cómo dice @dmarcosm es la oportunidad para cambiar al nuevo marlin, y de paso hacerlo en todas las impresoras. Así los nuevos también aprenden esa parte.

GioSunrider, 2018-12-15:
En la última reunión actualice el firmware de Spicy por Marlin 1.1.9., el mismo que tengo yo en casa, añadidos los custom comanda y habilitado el cambio de filamento. Mañana subiré los archivos de configuración al repo.

GioSunrider, 2018-12-16:
Subidos archivos en 21e93f3 
Tienen el añadido de lenguaje ya que esta pensado para PLA y PETG, en vez de ABS y PLA.

closed 2018-12-16 by GioSunrider