asrob-uc3m/impresoras-asrob #90: Piezas para Geonosis!

dmarcosm, 2018-09-21:
Geonosis necesita piezas nuevas para mejorar su estructura! 

De momento solo son 2 piezas para los motores de X e Y, y ya veremos si necesitamos algo más adelante :) 
- [ ] Sujecciones motores X e Y: https://www.thingiverse.com/thing:2613150/files

GioSunrider, 2018-09-21:
Rehacer el ventilador de capa lo necesita, pero no me acuerdo que piezas eran.

dmarcosm, 2018-09-21:
No es este, es parecido. Pero me gusta más, porque no queda el ventilador tan vertical y no tiene tantas posibilidades de morir de un golpe: https://www.thingiverse.com/thing:2659499/files

GioSunrider, 2018-09-22:
Pero no es el mismo ventilador, esa pieza no valdría, nuestro ventilador el aire sale por un lateral no por toda la superficie

dmarcosm, 2018-09-23:
Quizá este: https://www.thingiverse.com/thing:2828421/files
(aunque lo de que el nuestro sea vertical no sería un problema porque hay una pieza-adaptador imprimible)

JorFru, 2018-09-23:
Opinión personal justificable:
Ese ventilador radial enano que trae la Tronxy y que estáis obcecados en
usar es una absoluta basura.
Merece la pena gastarse 4 pavos en un ventilador nuevo. Un 40mm axial, o
uno radial más grande, como el de la A8.

Saludos xD

dmarcosm, 2018-09-23:
Me suena que en la última compra vino un ventilador centrífugo, más grande que el que tiene ya geonosis, pero que me confirme @Siotma

Siotma, 2018-09-27:
Efectivamente, compramos dos: #53

GioSunrider, 2018-09-27:
Compraria otro compatible con el nuevo extrusor que se instale en Blacky y pondría uno en Geonosis, como lo veis?

dmarcosm, 2018-09-27:
No entiendo tu pregunta @GioSunrider. Si ya tenemos 2 comprados, y uno va para blacky, ya nos sobra uno para geonosis, no?

dmarcosm, 2018-09-28:
Ah, vale. No lo había pensado :+1:

Siotma, 2018-10-13:
Con uno de los ventiladores que tenemos, se podría plantear algo como esto: https://www.thingiverse.com/thing:2819859 Yo propongo ponerlo primero solamente en un lado, porque ahora solo tenemos uno. Si para cuando llegue el envío de AliExpress vemos que hace falta uno en cada lado, se puede poner otro.

Siotma, 2018-10-13:
Alguien se ha puesto ya con este reto?

Siotma, 2018-10-27:
En la última reunión, Hija imprimió las dos sujecciones de motores:

[image]

Respecto a los ventiladores de capa, @jorgercobo y yo estuvimos dándole vueltas. Hemos pensado en volver a colocar las piezas originales y poner una tobera por dentro para dirigir mejor el aire al disipador y que no pegue en el hotend. Cuando tengamos claro cómo debe hacerse, abriremos otro reto de impresión.

closed 2018-10-27 by Siotma