asrob-uc3m/impresoras-asrob #91: Tampones para sello Feria de Asociaciones

JorFru, 2018-09-24:
Necesitamos tener impreso, para disponer de ello el miércoles, el siguiente item que consiste en un tampón para sello con el logo de ASROB. Se usará para validar a la gente que pase por el stand de ASROB en la gymkhana.
La pieza no tiene el tamaño correcto todavía, hay que escalarlo.
2-3cm de ancho, con 3cm de altura para que se pueda coger con la mano sin mancharse de tinta, a ser posible.
[ASROBstamp.zip](https://github.com/asrob-uc3m/impresoras-asrob/files/2410172/ASROBstamp.zip)

También he preparado otro sello, de regalo, para obtener más puntos en el reto de impresión. Está dentro del zip.

Hoy a mediodía voy a llevar filamento flexible de FFF color "azul topacio" para que se imprima.

GioSunrider, 2018-09-24:
Lo imprimo a la hora de comer en casa en PLA, a la hora de comer, y lo llevo esta tarde.

GioSunrider, 2018-09-25:
El sello quedo en el armario listo para su uso, se colgara posteriormente foto de él.

JorFru, 2018-09-25:
Yo tengo uno en filaflex, pero no quedó nada bien la primera capa.
Lo probaremos.

Siotma, 2018-09-28:
Cómo fue este reto?

GioSunrider, 2018-09-29:
Falta la foto que no lo hicimos pero bueno, el plan chupa poca tinta pero sirvió para su cometido.

JorFru, 2018-10-01:
[image]
Imprimí esto, descuidando la máquina y la primera capa (la importante para el resultado final del sello) no quedó del todo bien.
Al volver a intentar imprimirlo tuve un atasco interesante.

El día D por problemas organizativos no pude llevar la pieza impresa a la feria.

L

JorFru, 2018-10-01:
@GioSunrider te mereces cerrar la Issue

closed 2018-10-01 by GioSunrider