asrob-uc3m/impresoras-asrob #95: OVM20 ANET edition para Geonosis

JorFru, 2018-10-03:
Staticboards nos ha regalado su última creación para que la probemos como betatesters. Conviene instalarla lo antes posible para ir reportando a Javier Loureiro (Staticboards) las sugerencias y bugs de la nueva placa.

La placa original de la Tronxy irá al armario. El MOSFET que actualmente lleva Geonosis se moverá a Spicy #95

dmarcosm, 2018-10-06:
Pues no sé. Ahora después digo a Javier que os meta a uno de los dos en el grupo de betatesters, que yo estoy hasta arriba ahora.

JorFru, 2018-10-06:
De cabeza entro en ese grupo.
No tengo tiempo
pero para hablar de electrónica estoy siempre disponible.

Siotma, 2018-10-06:
Puntualizar que no ha sido un regalo: https://github.com/asrob-uc3m/impresoras-asrob/issues/87

JorFru, 2018-10-16:
Componentes pedidos. Estarán aquí a partir del 28 de octubre, creo.

Siotma, 2018-11-03:
@JorFru , algún dato interesante del último día para que conste?

JorFru, 2018-11-03:
El proveedor se equivocó y me mandó AMS1117-3.3 en vez de AMS1117-5.0
Conseguiré los reguladores correctos para instalarlos el lunes o martes después de la OSHWDem

Siotma, 2018-11-20:
Parece que la placa quedó reparada en la última sesión. @JorFru trató de flashear Marlin 1.1.9 sin éxito.

Hoy he estado investigando. He conseguido instalar la placa de la Anet en Arduino IDE siguiendo el tutorial: https://trello.com/c/QRIYELJA/311-configurar-arduino-ide-para-actualizar-el-firmware-a-partir-de-c%C3%B3digo-fuente

También he editado los archivos de configuración (adjuntos).

Sin embargo, no he sido capaz de subir ningún firmware a la placa porque parece que no tiene bootloader. @dmarcosm dice que puede ocurrir que al programarla se borre, así que igual eso fue lo que le ocurrió a @JorFru . Habrá que usar un ISP, en Arduino tenemos este el USBTinyISP. En el siguiente enlace aparecen los esquemáticos de los conectores: https://cdn-learn.adafruit.com/downloads/pdf/usbtinyisp.pdf

Según @dmarcosm hay que conectarlo así.
[image]

[Configuration_Geonosis.zip](https://github.com/asrob-uc3m/impresoras-asrob/files/2601415/Configuration_Geonosis.zip)

GioSunrider, 2018-12-15:
Me gustaría que actualizados el estado de la placa, si esta es funcional ya mismo, podéis proceder a cerrar la issue

GioSunrider, 2018-12-20:
Siguiendo el tutorial de staticsboards, que @Siotma proporciono, logre subir con exito el marlin 1.1.9, aunque los fichero que ha y en el repo de @dmarcosm estan mal configurados, ya que la pantalla ni la placa eran las correctas. 
Cuando se fue a comprobar la placa ocurrio lo siguiente....

[image]

Se rompio el conector usb, por lo uqe planteo dos soluciones al problema.
1. Si @JorFru puede repararlo, tiene via libre para hacerlo ya que la placa podemos actualizarla via programador, pero, no podemos conectarla a octoprint.
2. Retornar a la paca original de TronXY

JorFru, 2018-12-28:
Creo que puedo hacerlo.
Me propongo como plazo las dos primeras semanas de Enero.

Si necesitáis la máquina con el USB en funcionamiento antes de este periodo, podéis hacer lo que queráis, pero no me puedo comprometer a hacerlo en una fecha concreta.

GioSunrider, 2018-12-28:
No, no ha prisa, es más SAV, ya que todavía no hemos visto a geonosis quemando plástico. Pero no hay problema en tenerla parada, hay otras 4 Impresoras. Así que por mi parte tienes vía libre para ponerte con ella cuando puedas, te dejo encargado de esta issue para que hagas lo que creas mejor.

Siotma, 2019-01-13:
> 
> 
> Siguiendo el tutorial de staticsboards, que @Siotma proporciono, logre subir con exito el marlin 1.1.9, aunque los fichero que ha y en el repo de @dmarcosm estan mal configurados, ya que la pantalla ni la placa eran las correctas.

Según dices, utilizaste las fuentes de @dmarcosm , sin embargo, yo te había dejado un zip con la configuración en [mi post](https://github.com/asrob-uc3m/impresoras-asrob/issues/95#issuecomment-440414586). Supongo que no lo viste. Estuve configurándolo un buen rato y busccando información para asegurarme de lo que hacía... debería funcionar bien.

GioSunrider, 2019-01-13:
Pues la verdad que no lo vi, seria subir ese a ver si funciona, pero antes hay que arreglar la placa.

JorFru, 2019-01-26:
Jorfru ejecutó y @GioSunrider miró.
Siento no dejarte el placer de soldarla tron. La siguiente te la haces tú.

Estará disponible para montarse en la siguiente reunión.

[image]
[image]

GioSunrider, 2019-01-30:
Se ha intalado y subido el marlin que @Siotma configuro, funciona perfectamente menos por el LCD que ha y que configurarlo.

GioSunrider, 2019-01-30:
Sigo sin poder arreglarlo, para poder subir el marlin hay poner la placa tipo Anet V1.0 (Optiboot) para que se suba correctamente.

[image]

JorFru, 2019-01-31:
Yo en su día también hice muchos intentos para poner a funcionar esa pantalla LCD y no lo conseguí.

Siotma, 2019-02-14:
> 
> 
> Se ha intalado y subido el marlin que @Siotma configuro, funciona perfectamente menos por el LCD que ha y que configurarlo.

Mentira! Hoy @GioSunrider y yo hemos intentado hacer pruebas de movimiento con Geonosis y al hacer homing, el regulador de 5V HA VUELTO A ARDER! :tired_face: 

Necesitamos volver a colocar la placa antigua y poner la impresora a funcionar. Lleva casi un año en la asociación y ahora que tiene un V6 no puede estar parada...

JorFru, 2019-05-08:
Cuál es el estado actual?
La OVM se ha vuelto a freír? La reparo si hace falta.

Me da mucha pena ver esta máquina parada, sobre todo por el extrusor grueso que tiene y no le estamos dando uso.

MrTommyGuns, 2019-05-08:
Yep, sigue frita...
Yo le volvería a poner la placa anterior y pa lante a funcionar...
Si no me equivoco también tenemos placas donadas del hackatón. Podemos ver cuales rentan para cada impresora
Para nuestra impresora casta necesitariamos una plaquita de estas open supongo

GioSunrider, 2019-05-08:
Mi idea es poner la ovm 20 lite para la nueva a democratizar, una buena mecánica con una buena placa. La ovm anet sigue frita, quedamos en que habria que reparar y probarla una a una, y a geonosis ponerle su placa original.

GioSunrider, 2019-05-09:
@JorFru cambio de nuevo el regulador. Esta vez ardio nada mas encender. Hemos decidido quitar la placa y volver a la orginal. Ahora vuelve a funcionar y se estan haciendo n uevas pruebas de impresion.

GioSunrider, 2019-05-09:
Doy esta issue por cerrada dado que no se va a utilizar esa placa.

closed 2019-05-09 by GioSunrider

JorFru, 2019-05-09:
Pruebas de impresión satisfactorias. La placa antigua funciona correctamente, su firmware (Marlin 1.1.6) parece funcionar correctamente.