asrob-uc3m/impresoras-asrob #97: Ideas Proyectos - Subvenciones curso 2018-2019

dmarcosm, 2018-10-05:
Vayamos dejando aquí las ideas de proyectos que se nos vayan ocurriendo, ya que no tardando mucho tendremos que pedir las subvenciones de este curso (sabemos fechas @GioSunrider?)

GioSunrider, 2018-10-05:
Todavía no tengo ni idea, pero si, de momento una de mantenimiento de la granja de impresoras hay que pedir.

dmarcosm, 2018-10-05:
Como idea loca propongo montarnos una Prusa Blackbelt, es decir, añadirle una cinta motorizada a una de las impresoras tipo prusa que tenemos. El proyecto en sí no sería demasiado complicado, ya que sería, a grandes rasgos, usar el motor del eje Y para hacer girar una cinta de un material X alrededor de la cama caliente. Habría que ver si habría que aumentar el tamaño de la impresora por debajo para que cupiese el mecanismo, pero no lo veo muy complicado :)

GioSunrider, 2018-10-06:
Yo pediría dos subvenciones entonces, una de mantenimiento del parque de impresoras y otra de proyecto de mejora. En la de mejora podemos incluir perfilería y demás, así también tenemos dinero y margen de maniobra.

dmarcosm, 2018-10-06:
Lo de la AM8 lo veo muy factible, además no nos costaría demasiado. Quizás haría en general una subvención de "Creación y mejora de impresoras", y detallar dentro todo lo que queremos hacer

GioSunrider, 2018-10-06:
Yo lo decía por la compra de filamento! Qué cogería solo una subvención para la compra de varios tipos de filamento e investigar.

dmarcosm, 2018-10-06:
> Yo lo decía por la compra de filamento! Qué cogería solo una subvención para la compra de varios tipos de filamento e investigar.

Subvención para material? Tipo perfilería, filamento, mejoras de impresoras (aka autolevel, bases, etc..)

dmarcosm, 2018-10-06:
> Todavía no tengo ni idea, pero si, de momento una de mantenimiento de la granja de impresoras hay que pedir.

Realmente, qué hay que mejorar exactamente en la granja de impresoras? O sea, tenemos algo pensado?

GioSunrider, 2018-10-07:
Perfilería, y bueno la tronxy necesita una buena mano de muchas cosas, igual sería la que se llevaría todo el presupuesto.

dmarcosm, 2018-10-15:
Propongo también transformar una de las impresoras que tenemos con RAMPS a multicolor, con doble extrusor, lo cual sería muy interesante para los soportes solubles :)

dmarcosm, 2018-10-16:
- Cambio de extrusor de Dedalo, a 1.75mm
- Añadir laser engraver/cutter a una impresora (anet, por ejemplo)

GioSunrider, 2018-10-17:
Propongo aglutinar, una unica subvencion de "mejora y mantenimiento de la granja de impresoras." De esta manera tenemos margen para aplazar ideas y asi tambien dejamos subvenciones para el resto de la asociacion.

dmarcosm, 2018-10-17:
Yo pienso que deberíamos pedir al menos 2 subvenciones separadas. Por no generalizarla tanto y que luego quizá algo no podamos pasarlo por esa, pero el experto en subvenciones es @Siotma.

GioSunrider, 2018-10-17:
Hay que ver cuantas subvenciones se piden en general de asociacion, porque si pedimos todos, dos igual son muchas.

dmarcosm, 2018-10-17:
> Uno de los años (no recuerdo si el pasado o antes de eso) sí que estaba regulado el número máximo por asociación. Una posible "trampa" es pedirla a título individual como alumno, que también se puede.

En vez de a título personal, se podrían pedir en nombre del grupo? O tiene que ser asociación per se? De todas formas yo creo que no saldrán tantas peticiones para todo asrob, así que no me preocuparía.

David-Estevez, 2018-10-17:
O a título personal, o bajo una asociación.

GioSunrider, 2018-11-05:
Vale chicos, hay que saber que queremos pedir exactamente. @dmarcosm no te vengas arriba que hay que aglutinar todo lo que podamos, dado a que tiene pinta que vamos a estar limitados, también hay que pensar que nos dará tiempo a hacer menos de un año.

David-Estevez, 2018-11-05:
Como he sugerido por el grupo de RoboFactory, primero apuntad qué es lo que queréis pedir (si hubiese dinero/subvenciones ilimitadas) y a partir de eso trabajamos.

También estaría guay que hicieséis un desglose de las cosas que harían falta en cada proyecto y cuánto estimáis que costarían.

A partir de eso haremos nuestra magia de siempre.