asrob-uc3m/impresoras-asrob #101: Alimentacion de RPi's para la granja

GioSunrider, 2018-10-08:
Comprar cables microusb, [este pack de 5](https://www.amazon.es/MaGeek-Sincroniza-Samsung-Motorola-Colores/dp/B01A7BVDES/ref=sr_1_3?ie=UTF8&qid=1539023141&sr=8-3&keywords=cable+microusb+1m) seria suficiente y dos bases dobles como [este pack](https://www.amazon.es/Cargador-Luvfun-Tecnolog%C3%ADa-tel%C3%A9fonos-inteligentes/dp/B07FX8434B/ref=sr_1_17?ie=UTF8&qid=1539023304&sr=8-17&keywords=cargador%2Busb&th=1)

Siotma, 2018-10-11:
Buena opción!

dmarcosm, 2018-10-12:
Vería más lógico comprar una base con 4 USBs https://www.google.es/aclk?sa=L&ai=DChcSEwjrlfPlu4DeAhUJzncKHRqbAXUYABAEGgJlZg&ae=1&sig=AOD64_3Yh1je03cbveEalXK1rfxOz9z9Ww&ctype=5&q=&ved=0ahUKEwjYxuvlu4DeAhUC66QKHd3QDZcQwg8ILA&adurl=https://www.amazon.es/AUKEY-Cargador-Tecnolog%25C3%25ADa-Tabletas-Dispositivos/dp/B00TD3Y4ZM/ref%3Dasc_df_B00TD3Y4ZM/%3Ftag%3Dgoogshopes-21%26linkCode%3Ddf0%26hvadid%3D54376939155%26hvpos%3D1o2%26hvnetw%3Dg%26hvrand%3D3548963071788184925%26hvpone%3D%26hvptwo%3D%26hvqmt%3D%26hvdev%3Dm%26hvdvcmdl%3D%26hvlocint%3D%26hvlocphy%3D9061032%26hvtargid%3Dpla-151532731275%26psc%3D1, así no gastariamos tantos enchufes.

Siotma, 2018-10-12:
Las raspis puede que estén separadas en dos baldas, como lo están ahora mismo las impresoras, por lo que parece interesante tener dos adaptadores distintos. Además la opción de dos bases es más barata que la de una base.

Siotma, 2018-10-12:
Acabo de hacer el pedido de la propuesta de @GioSunrider

GioSunrider, 2018-10-12:
Genial!

Siotma, 2018-10-16:
Hoy he llevado a la nave los adaptadores de 5V, y también he recibido los cables en mi casa:

[image]
[image]

closed 2018-10-16 by Siotma