asrob-uc3m/impresoras-asrob #102: Comprar nuevos filamentos

Siotma, 2018-10-08:
Para apurar el presupuesto de las subvenciones, @GioSunrider y yo hemos pensado en pedir bobinas de filamento. Se aceptan sugerencias.

JorFru, 2018-10-08:
Yo quiero esto papi
https://www.materials3dimpresion.com/pla_ep.php?l=0

dmarcosm, 2018-10-08:
- https://fillamentum.com/products/pla-extrafill-vertigo-grey o https://fervi3d.com/fiberlogy-easy-pla/335-722-fiberlogy-easy-pla-vertigo.html
- https://smartmaterials3d.com/es/tienda-smart/20-smartfil-flex.html#/diametro-1_75_mm_0_03/color-ruby/tamano-s_330g
- https://smartmaterials3d.com/es/smartfil-pla/72-smartfil-pla-mix.html 
No se cuanto dinero hay pero quizá nos interesa probar el Nylon: https://fervi3d.com/fiberlogy-nylon-pa12/448-903-nylon-pa12-black.html#/25-diametro-175_mm/28-tamano-m_750_g

GioSunrider, 2018-10-09:
Para variar un poco de filamentos yo propongo PETG de Fiberlogy que tiene muy buena calidad. https://www.3dpenedes.com/es/filamento/973-pet-g-azul-transparente.html

dmarcosm, 2018-10-09:
Cuánto hay sobrante @Siotma?

GioSunrider, 2018-10-09:
No era mucho, unos 60 creo recordar, pero lo de los filamentos lo dejaremos para la ultima semana de octubre no vaya a ser que necesitemos algo antes y no podamos.

Siotma, 2018-10-11:
Tienen todos muy buena pinta.

Una vez veamos el dinero disponible, realizaré la compra :wink:

Siotma, 2018-10-19:
Esta cesta de la compra liquidaría las subvenciones:

[image]

dmarcosm, 2018-10-19:
> Esta cesta de la compra liquidaría las subvenciones:
> 
> [image]

A mi me parece correcto :)

GioSunrider, 2018-10-20:
Se que no viene a cuento pero... y algo de estaño? en general estamos muy necesitados de el, se podria incluir?

dmarcosm, 2018-10-20:
La cosa es que el filamento es algo que hay qee comprar vía la issue de material común, no? De hecho creo que ya está puesto allí.

GioSunrider, 2018-10-20:
Me refería para terminar de cuadrar las cuentas.

dmarcosm, 2018-10-20:
Pero habría que quitar algún filamento entonces, porque con lo poco que "sobra" no compras demasiado estaño xD 
(si hay que quitar alguno yo pasaría del Smart Glow)

GioSunrider, 2018-10-20:
Yo quitaría el smart glow, además para así comprar kits para el ajuste del extrusor para las otras impresoras.

JorFru, 2018-10-20:
Del estaño me encargo yo. Dedicad el presupuesto de Printer para filamentos y componentes.

Siotma, 2018-11-07:
Hoy se han comprado 4 filamentos:
- https://fervi3d.com/smartfil-flex-93a/13-89-flex.html#/25-diametro-175_mm/27-tamano-s_330_g/34-color-sapphire
- https://fervi3d.com/fiberlogy-petg/330-717-fiberlogy-petg-tr-navy-blue.html#/25-diametro-175_mm/98-tamano-xm_850_g
- https://fervi3d.com/esencial-pla/406-790-pla-fervi3d-esencial.html#/25-diametro-175_mm/28-tamano-m_750_g/105-color_esencial-verde_luminoso
- https://fervi3d.com/fiberlogy-easy-pla/444-895-fiberlogy-easy-pla-onix.html#/25-diametro-175_mm/98-tamano-xm_850_g

Siotma, 2018-11-18:
Los filamentos llegaron hace unos días. Los llevaré en la próxima reunión:

[image]

closed 2018-11-18 by Siotma