asrob-uc3m/impresoras-asrob #98: Instalación Octorpint

GioSunrider, 2018-10-08:
Arrastrado de #49. Se necesita la instalacion de Octoprint para dar salida con 3 RPi's y 5 camaras para el control de las impresoras.
Los pasos a seguir son:

- [ ] Clonar la RPi que @dmarcosm tiene con dos estancias.

- [ ] Imprimir el [Protector de la RPi 3B](https://www.thingiverse.com/thing:922740

- [ ] Creacion de 5 bots de Telegram con el nombre de la Impresora

- [ ] Añadir los bots al grupo de telegram de [impresoras](https://telegram.me/joinchat/ADT1EAoc4Ds9nBXp4Hj9dw)

- [ ] Colocacion de la RPi

dmarcosm, 2018-10-08:
En cuanto pueda subo los archivos o una guía para no retrasar esto más. Also, propongo no imprimir una caja para cada raspberry, y en su lugar hacernos/diseñarnos una caja en la que meter las 3 raspberries. Podríamos hacer un taller para aprender freecad haciendo la caja (@siotma?), o que alguien se encargue. Ahí dejo la propuesta

GioSunrider, 2018-10-08:
Dado que las Rpi's están colocadas en distintos lugares mejor separadas que juntas.

GioSunrider, 2018-10-08:
Arrastrado tambien de #65

dmarcosm, 2018-10-08:
> Dado que las Rpi's están colocadas en distintos lugares mejor separadas que juntas.

Yo lo decía sobre todo por no tener ahí 3 raspberries sueltas. Siendo la intención cerrar todo (caja de filamentos, cerrar granja), vería muy lógico cerrar todas las Pi's en una caja, ya que además no vamos a tener que tocarlas para nada.

GioSunrider, 2018-10-08:
Lo dejaría de plan a futuro, seguro que esas carcasas se pueden reutilizar en un futuro. Pero de momento busco la rapidez de instalación.

David-Estevez, 2018-10-08:
> Podríamos hacer un taller para aprender freecad haciendo la caja (@Siotma?), o que alguien se encargue. Ahí dejo la propuesta

Me ha contado un pajarito que en el GUL ante este tipo de peticiones hacen que la persona que lo pide dé el taller... dejo ahí la idea... :thinking:

dmarcosm, 2018-10-08:
> > Podríamos hacer un taller para aprender freecad haciendo la caja (@Siotma?), o que alguien se encargue. Ahí dejo la propuesta
> 
> Me ha contado un pajarito que en el GUL ante este tipo de peticiones hacen que la persona que lo pide dé el taller... dejo ahí la idea...

Yo no tengo problema alguno en aprender Freecad para hacer un taller :D 

> Lo dejaría de plan a futuro, seguro que esas carcasas se pueden reutilizar en un futuro. Pero de momento busco la rapidez de instalación.

Llevamos meses sin hacer octoprint y ahora tenemos prisa en ponerlas en una caja? Hasta que no se adjudiquen en febrero las nuevas subvenciones tenemos bastantes cosas para hacer, así que no veo la prisa en ningún lado.

Si os apetece el taller lo podemos preparar (algo no muy advanced)

David-Estevez, 2018-10-08:
Si no, siempre quedan los tutos de Obijuan: https://www.youtube.com/watch?v=2_DbFzFV9D4&list=PLmnz0JqIMEzWQV-3ce9tVB_LFH9a91YHf

GioSunrider, 2018-10-15:
Durante el puente me traje las dos RPi a casa y siguiendo el tutorial que @dmarcosm realizo ([Ver Enlace](https://github.com/asrob-uc3m/impresoras-asrob/blob/master/octoprint/guia_segunda_instancia.md)), he creado dos estancias en ambas Pi's. Ademas de actualizar y corregir los errores encontrados en el proceso y añadir algunos apartados en el mismo.

dmarcosm, 2018-10-15:
Como dato, en un futuro no muy lejano se intentará hacer todo en un solo script que el usuario pueda ejecutar con un solo comando, así podremos publicarlo y que lo use quien quiera :)
Sobre la instalación, es posible que vaya el jueves, así que si no queda hecho en la reunión y dejas las Pi's allí me encargo @GioSunrider

GioSunrider, 2018-10-15:
En el armario se quedaran. Mañana comenzaremos la instalacion de una de ellas, ya que es la unica que tendria carcasa.

dmarcosm, 2018-10-15:
Cuidado con lo de añadir los 5 bots al grupo. Se puede crear un spam rico rico.

GioSunrider, 2018-10-15:
La idea es usar el grupo que hay de impresoras 3D, no el de printer, de control de Octoprint. Deshabilitar las tajetas SD por Marlin, de esta manera se verifica el uso por parte de un operador, ademas de restringir su uso local y ternerlas controladas por ahi los admins.

dmarcosm, 2018-10-15:
> La idea es usar el grupo que hay de impresoras 3D, no el de printer, de control de Octoprint. Deshabilitar las tajetas SD por Marlin, de esta manera se verifica el uso por parte de un operador, ademas de restringir su uso local y ternerlas controladas por ahi los admins.

Me parece correcto, pero no del todo. Te quedas sin posibilidad de imprimir offline.

Siotma, 2018-10-16:
Las ideas que habéis dado están muy bien, pero no hay que perder de vista que la granja necesitará un tiempo de puesta a punto durante el cual no se podrá dejar las impresoras sin SD, hasta que toda la infraestructura de octoprint:
- [ ]  funcione
- [ ] se haya documentado y 
- [ ] se haya formado a la gente para usarla.

El sistema que mencionais, por tanto, es a lo que hay que tratar de llegar, pero no es un futuro cercano. Hay que ponerlo a prueba para encontrar la mejor configuración.

GioSunrider, 2018-10-17:
Ayer se imprimio con hija una de las cajas para las RPi's. De esta manera han quedado ya dos debidamente protegidas con carcasa (Ver Foto).

[image]

Faltaria por tener una tercera carcasa para la que tiene @dmarcosm en su casa. Lo propongo como RETO DE IMPRESION, se puede dividir en dos personas, para tenerla como esta a dos colores, ¿Algun voluntario?

GioSunrider, 2018-10-27:
En la ultima reunion coloque las RPi en la granja

![photo5954216383850065662](https://user-images.githubusercontent.com/10605193/47605850-
49f42580-da0c-11e8-9a60-7a945df80286.jpg)

 y crimpe los cables necesarios a medida para dotar de internet a la granja.

[image]
 
Ademas @anacarnerero diseño una pieza para la colocación del router.

Siotma, 2018-10-28:
Imprimiendo parte de arriba de la caja (top) en color negro.

GioSunrider, 2018-10-30:
Imprimiendo parte de abajo de la caja (Bottom) en color blanco

GioSunrider, 2018-11-02:
En la pasada reunion se termino la colocacion de las RPi como se puede observar en la foto la colocacion. 

[image]

Ademas se conectaron todos los cables hechos a medida y su correspodiente alimentacion. Solo faltaria la colocacion de las Camaras y unir con el cable impresora y RPi.

Ademas se completo la parte de red conectando y colocando el router que dara soporte a la granja, en el perto de Ethernet libre se conecto la torre del ordenador. Se adjunta foto del router colocado. para ello @anacarnerero diseño una pieza que se imprimio, despues de hacerle algun ajuste se coloco y cableo.

[image]

GioSunrider, 2018-11-05:
Se han creado y añadido los bots al grupo de gestion de las impresoras. Para su creacion se ha determinado median el (Nombre de la impresora)_Printer_bot.

GioSunrider, 2018-11-05:
Para el proximo dia se añadiran a Octoprint y se realizara la gestion de la informacion que envien al grupo.

GioSunrider, 2018-11-15:
Propongo [este soporte](https://www.thingiverse.com/thing:2587839) para hija, blacky y spicy. Para Dédalo y para genoosis habría que buscar alternativas.

GioSunrider, 2018-11-27:
He añadido un edit al comentario inicial, para laconfiguracion de las camaras hay que seguir el tutorial de configuracion proporcionado por @dmarcosm en la siguiente [pagina web](http://thomas-messmer.com/index.php/14-free-knowledge/howtos/79-setting-up-octoprint-for-multiple-printers) y traducirlo para añadirlo a nuestro tutorial. Me pondre esta tarde en la reunion con ello.

GioSunrider, 2018-11-27:
La RPi de Spicy y Geonosis queda asignada con una IP estatica http://192.168.0.100/ y http://192.168.0.100:5001/ respectivamente.

jorgercobo, 2018-12-13:
Se han etiquetado las RPi que controlan cada impresora:
- __100__: 
  - Spicy: http://192.168.0.100/ 
  - Geonosis: http://192.168.0.100:5001/
- __101__: 
  - Dedalo
- __102__: 
  - Hija Rs
  - Blacky

jorgercobo, 2018-12-13:
El tty al que se conectan las impresoras está configurado de forma automática y puede que los puertos no se mantegan al reiniciar la RPi, además es necesario para la configuración de las webcams fijar el tty de cada impresora. Para ello hay que buscar los ttyUSB que ahora no aparecen, para ello voy a seguir [esta guia](https://forum.level1techs.com/t/solved-octoprint-help-no-serial-no-ttyusb/110536/21). Estoy haciendo estas pruebas en 102.

jgvictores, 2018-12-13:
Te recomendaría:
1. No demasiado caso al tutorial, salvo que quizás ayude el `sudo modprobe ftdi-sio`
1. Un `tail -f /var/log/syslog` para ver qué ocurre cuando se conectan los dispositivos físicamente (parecido a [dmesg](https://github.com/asrob-uc3m/tutoriales/blob/master/software/linux/bash.md#listado-alfab%C3%A9tico-de-comandos-%C3%BAtiles)).
1. Lo que más me preocuparía es que al tener el las impresoras el mismo adaptador, no se controla qué ttyUSB es cuál. Aquí una posible solución, que los diferencia en base a qué conector se conectan físicamente: https://stackoverflow.com/questions/5421008/how-to-uniquely-identify-two-identical-usb-serial-adapters-in-linux (related: https://github.com/asrob-uc3m/impresoras-asrob/issues/2#issuecomment-283737116)

jorgercobo, 2018-12-13:
Muchas gracias. El [tutorial](http://thomas-messmer.com/index.php/14-free-knowledge/howtos/79-setting-up-octoprint-for-multiple-printers) que estamos siguiendo para configurar todo tambien usa ese método. Para la siguiente reunión queda pendiente seguir con lo que sugiere @jgvictores. 
En 102: 
  - Pendiente identificar cada impresora mediante udevd. Por ahora se hace automático y parece funcionar, pero aseguremonos.
  - Las reglas para las webcams están configuradas y los daemons duplicados y corriendo.
  - Falta por hacer:
    - HaProxy Configuration
    - Tras la parte de las webcams: Another steps follows :) ... We have to make the new instance visible behind HaProxy.

jorgercobo, 2018-12-19:
- HaProxy ya está configurado, a traves de http://192.168.0.102/webcam/ y http://192.168.0.102/webcam2/ se puede acceder a ambas camaras. En webcam se ve la ventana de error de webcams de octoprint como se ha configurado y en webcam2 se ve la pantalla de control de mjpg_streamer. Queda pendiente estudiar las reglas de HaProxy para poder acceder a las webcams mediante /webcam... independientemente de la URL de origen ya que ahora hay que forzar la URL absoluta en Hija al fallar debido al puerto 5001 en la URL.
- El problema de las webcams persiste. Algo que antes no habiamos mencionado es que el led de una de las camaras no se enciende. Esto ocurre tanto en 100 como en 102, **¿puede ser un problema de alimentación?**. Mediante lsusb aparecen ambas webcams pero al ejecutar `tail -f /var/log/syslog` se ve el siguiente mensaje de error de forma periódica:
    Dec 19 17:34:44 octopi pi: Starting USB webcam
    Dec 19 17:34:44 octopi MJPG-streamer [5627]: starting application
Dec 19 17:34:44 octopi MJPG-streamer [5627]: MJPG Streamer Version: git rev: 821c330ea6bbb5fbed98d48e00aac156e923161b
Dec 19 17:34:44 octopi MJPG-streamer [5627]: Using V4L2 device.: /dev/video0
Dec 19 17:34:44 octopi MJPG-streamer [5627]: Desired Resolution: 640 x 480
Dec 19 17:34:44 octopi MJPG-streamer [5627]: Frames Per Second.: 10
Dec 19 17:34:44 octopi MJPG-streamer [5627]: Format............: JPEG
Dec 19 17:34:44 octopi MJPG-streamer [5627]: TV-Norm...........: DEFAULT
Dec 19 17:34:44 octopi MJPG-streamer [5627]: init_VideoIn failed

Tambien adjunto el estado de cada servicio por si aporta info:

pi@octopi:/etc/init.d $ systemctl status webcamd.service
● webcamd.service - LSB: webcam daemon
   Loaded: loaded (/etc/init.d/webcamd; generated; vendor preset: enabled)
   Active: active (running) since Thu 2018-12-13 16:17:04 UTC; 6 days ago
     Docs: man:systemd-sysv-generator(8)
   CGroup: /system.slice/webcamd.service
           ├─ 454 /bin/bash /root/bin/webcamd
           └─5406 sleep 30

Dec 19 17:28:13 octopi mjpg_streamer[5317]: MJPG-streamer [5317]: Format............: JPEG
Dec 19 17:28:13 octopi mjpg_streamer[5317]: MJPG-streamer [5317]: TV-Norm...........: DEFAULT
Dec 19 17:28:43 octopi pi[5404]: Starting USB webcam
Dec 19 17:28:43 octopi mjpg_streamer[5405]: MJPG-streamer [5405]: starting application
Dec 19 17:28:43 octopi mjpg_streamer[5405]: MJPG-streamer [5405]: MJPG Streamer Version: git rev: 821c330ea6bbb5fbed98d48e00aac156e9231
Dec 19 17:28:43 octopi mjpg_streamer[5405]: MJPG-streamer [5405]: Using V4L2 device.: /dev/video0
Dec 19 17:28:43 octopi mjpg_streamer[5405]: MJPG-streamer [5405]: Desired Resolution: 640 x 480
Dec 19 17:28:43 octopi mjpg_streamer[5405]: MJPG-streamer [5405]: Frames Per Second.: 10
Dec 19 17:28:43 octopi mjpg_streamer[5405]: MJPG-streamer [5405]: Format............: JPEG
Dec 19 17:28:43 octopi mjpg_streamer[5405]: MJPG-streamer [5405]: TV-Norm...........: DEFAULT

pi@octopi:/etc/init.d $ systemctl status webcamd2.service
● webcamd2.service - LSB: webcam2 daemon
   Loaded: loaded (/etc/init.d/webcamd2; generated; vendor preset: enabled)
   Active: active (running) since Thu 2018-12-13 16:17:04 UTC; 6 days ago
     Docs: man:systemd-sysv-generator(8)
   CGroup: /system.slice/webcamd2.service
           ├─449 /bin/bash /root/bin/webcamd2
           └─508 ./mjpg_streamer -o output_http.so -w ./www-octopi -p 8081 -i input_uvc.so -d /dev/videoBLACKY -r 640x480 -f 10 -d /dev

Dec 13 16:17:05 octopi mjpg_streamer[508]: MJPG-streamer [508]: Frames Per Second.: 10
Dec 13 16:17:05 octopi mjpg_streamer[508]: MJPG-streamer [508]: Format............: JPEG
Dec 13 16:17:05 octopi mjpg_streamer[508]: MJPG-streamer [508]: TV-Norm...........: DEFAULT
Dec 13 16:17:05 octopi mjpg_streamer[508]: MJPG-streamer [508]: www-folder-path......: ./www-octopi/
Dec 13 16:17:05 octopi mjpg_streamer[508]: MJPG-streamer [508]: HTTP TCP port........: 8081
Dec 13 16:17:05 octopi mjpg_streamer[508]: MJPG-streamer [508]: HTTP Listen Address..: (null)
Dec 13 16:17:05 octopi mjpg_streamer[508]: MJPG-streamer [508]: username:password....: disabled
Dec 13 16:17:05 octopi mjpg_streamer[508]: MJPG-streamer [508]: commands.............: enabled
Dec 13 16:17:05 octopi mjpg_streamer[508]: MJPG-streamer [508]: starting input plugin input_uvc.so
Dec 13 16:17:05 octopi mjpg_streamer[508]: MJPG-streamer [508]: starting output plugin: output_http.so (ID: 00)

jorgercobo, 2018-12-20:
Pendiente de bajar la resolución de las camaras o los fps ya que el problema puede ser de ancho de banda

GioSunrider, 2019-11-15:
En la reunion de ayer se movio el router cerca del ordenador y se instalo en su lugar un switch de 8 puertos que dara servicio a la granja de impresoras. Haria falta imprimir un soporte para que ambos queden fijos. Ademas haria falta un Rpi mas que de servicio a la nueva impresora ademas de Spicy.

GioSunrider, 2019-11-28:
foto de la instalacion del switch

[image]

GioSunrider, 2019-11-28:
Se necesita terminar de configurar la segunda estancia de RPi que controla a Dedalo y Geonosis.
Se ha bajado la calidad de las Webcams que controlan a la Moebyus One y a Spicy.
Se ha conseguido que la camara de Blacky funcione, queda revisar el resto.

closed 2023-09-28 by CarlosME17