asrob-uc3m/impresoras-asrob #103: Comprar muelles para las camas calientes

Siotma, 2018-10-11:
Desgraciadamente, algunas impresoras, como Hija Resurrection, tienen muelles lamentables en la cama caliente, que dificultan su calibración y mantenimiento. Propongo comprar nuevos muelles para todas las impresoras que los necesiten: https://fervi3d.com/mecanica/188-muelles-cama-calienta-pack-4unidades.html

dmarcosm, 2018-10-16:
Muelles, tambien en Fervi3d: https://fervi3d.com/mecanica/188-muelles-cama-calienta-pack-4unidades.html 

1,50€ por cada 4 unidades :+1:

Siotma, 2018-11-07:
Hoy se han comprado 3 kits de 4 muelles en https://fervi3d.com/mecanica/188-muelles-cama-calienta-pack-4unidades.html

Siotma, 2018-11-18:
Los muelles llegaron hace unos días. Los llevaré en la próxima reunión:

[image]

closed 2018-11-18 by Siotma