asrob-uc3m/impresoras-asrob #105: Mejora de Extrusor de Spicy

GioSunrider, 2018-10-14:
Actualmente tengo una pieza para un extrusor V6 de sobra, ademas de una sujecion de correas. Como en #85 se compraron extrusores de sobra, propongo la implementacion de el sistema Bowden para spicy. Este sistema mejoraria la calidad de las impresiones, ademas de reducir esfuerzo mecanico a los motores.

dmarcosm, 2018-10-14:
No creo que sea lo idóneo. Hemos planteado comprar/pedir subvención para experimentar con nuevos filamentos. Uno de los filamentos que muy probablemente probemos sea flexible. Flexible ≠ bowden. Dejaría al menos una impresora con extrusor directo no sólo por el motivo de cuando investiguemos con flexible, si no porque quizá alguien quiere imprimir con flexible. Todo bowden no :(

GioSunrider, 2018-10-14:
mi idea era dejar en extrusor directo a blacky por varios motivos:
1. Espacio, el marco creo que no es el idoneo para bowden.
2. Es del departamento.
3. Por los motivos de poder imprimir flexible

Ademas tambien dejaria en directo a dedalo, luego ya serian dos impresoras que tendriamos en directo.

Siotma, 2018-10-16:
De acuerdo con que usen directo Blacky + Spicy. Había oído que para Anet hay una pieza mágica que permite imprimir con Filaflex no?

Siotma, 2018-10-19:
Teniendo en cuenta lo bien que imprimen las ANET (o al menos siempre habeis dicho eso), yo trataría de perfeccionar su extrusor antes de dar por hecho que lo vamos a cambiar. Hay ANETs que imprimen en flexible con el extrusor stock + la pieza que dice @dmarcosm . Yo creo que deberíamos intentar que Spicy fuera en esa dirección.

Spicy está nueva. Si imprime bien en su estado actual, cambiarle el extrusor es un riesgo de que empiece a fallar. Solo debe hacerse en caso de que ninguna otra impresora lo necesite más, y yo creo que no es el caso.

Además, no se puede dar por hecho que al ponerle Bowden la calidad va a aumentar, porque el sistema es más complejo y puede dar fallos.

Siotma, 2018-10-27:
En la última reunión se intentó imprimir con Spicy con PLA y se vio que el engranaje que mueve el filamento hacia el extrusor da saltos. Esto provocó que no se pudiera imprimir con Spicy. Invoco a los dioses de las ANET para tratar de solventar este problema. Caso de no ser posible, propongo cambiar el hotend completo por uno de los nuevos, previo acuerdo con @JorFru

dmarcosm, 2018-10-27:
A qué temperatura estaba imprimiendo, por curiosidad?

JorFru, 2018-10-28:
Antes de decidir el cambio de piezas, se debería de investigar mejor las causas del problema.
Lo primero que se me ocurre es reducir la distancia de retracción. Cuando retrae, el filamento se enfría, y cuando vuelve a empujar no fluye porque está frío. 2mm debería de ser suficiente para extrusión directa.
Todos los problemas que yo he tenido con ella en el extrusor han sido por distancia de retracción incorrecta.

Siotma, 2018-10-31:
De acuerdo @JorFru , seguiremos investigando.

GioSunrider, 2018-12-15:
Realizado el cambio en #120 esto ya no es necesario ya que no ha vuelto a haber problemas al respecto. Procedo a cerrar la discusión.

GioSunrider, 2018-12-15:
También debido a #107 en la que se acordó la colocación de los extrusores.

closed 2018-12-15 by GioSunrider