asrob-uc3m/impresoras-asrob #106: Actualización general a Marlin 1.1.9

dmarcosm, 2018-10-15:
Como hemos dicho en #104, es hora de actualizar todas nuestras impresoras a Marlin 1.1.9, la última versión estable. 
He dejado los archivos de todas las impresoras menos Spicy ya actualizadas. Añadir que aun después de arreglar la OVM20 de Geonosis, no se debe instalar el firmware que hay subido, ya que queda cambiar cosas como el modelo de la placa y los motores de X e Y (@JorFru) 
Por otra parte, necesito @Siotma que me confirmes si sabes si los drivers que utilizan Dédalo, Blacky e Hija son A4988 o DRV8825. Marlin por defecto coge los A4988, así que si alguna placa usa los DRV habrá que cambiarlo antes de subir nada (no queremos humo!)

Siotma, 2018-10-16:
Gran trabajo David. Estamos en proceso de actualizaciones.

Por cierto, Dédalo lleva DRV8825 via https://www.staticboards.es/productos/ovm20-lite/
Por favor, indícanos qué cambios hay que hacer para que los soporte...

Siotma, 2018-10-16:
Blacky e Hija llevan A4998 via https://www.pccomponentes.com/bq-kit-de-electronica-ramps-1-4

dmarcosm, 2018-10-16:
> Gran trabajo David. Estamos en proceso de actualizaciones.
> 
> Por cierto, Dédalo lleva DRV8825 via https://www.staticboards.es/productos/ovm20-lite/
> Por favor, indícanos qué cambios hay que hacer para que los soporte...

Ya ha quedado hecho, pero para la próxima aquí podéis ver qué cambios hay que hacer. Es simplemente definir el modelo que estemos usando: https://github.com/asrob-uc3m/impresoras-asrob/commit/efa813dfe8ea802819b449c747f9eb1dba94f0f2

Siotma, 2018-10-16:
Blacky actualizada. Hemos modificado los archivos via 6c78d1ae91d9864f3c4328e8c9b3b81d6aa9a9ba
- El archivo Configuration.h tenía los endstop invertidos. Lo hemos invertido a mano.
- Hemos habilitado el custom menu.

Siotma, 2018-10-16:
Dedalo actualizada. Hemos modificado los archivos via 506d392906b08315b79a63e381e4fb5b81fca3f7
- Al archivo Configuration.h le faltaban líneas de tipo //DEDALO en los DRV.
- Hemos habilitado el custom menu.

dmarcosm, 2018-10-16:
> Hemos habilitado el custom menu.

Esto es...?

GioSunrider, 2018-10-16:
> > Hemos habilitado el custom menu.
> 
> Esto es...?

Aceso directo a los principales comandos, como cooldown, prepare y demas, aunque en realidad se le puede poner lo que quieras.

Siotma, 2018-10-16:
Es lo que comenté en https://github.com/asrob-uc3m/impresoras-asrob/issues/84#issuecomment-425507793
Habilitando el custom menu se añade una opción nueva al menú principal, donde puedes añadir hasta 5 comandos personalizados. Lo ideal es poner ahí lo que más se usa para no tener que estar buscándolo entre los submenús, y además puede haber combinaciones de varias cosas. Todo lo que haya aquí está a 2 clics de la pantalla principal. Por ejemplo se puede configurar así:
- Precalentar + home.
- Enfriar.
- Home.
- Introducir filamento.
- Extraer filamento. 

En 506d392906b08315b79a63e381e4fb5b81fca3f7 y 6c78d1ae91d9864f3c4328e8c9b3b81d6aa9a9ba se puede comprobar los comandos que hemos puesto hoy. En un futuro, el uso dirá cuáles es más interesante tener.

GioSunrider, 2018-10-16:
Yo probaria el de cambio de filamento, mientras esta en impresion, igual conviene tenerlo mas a mano...

dmarcosm, 2018-10-16:
> Es lo que comenté en [#84 (comment)](https://github.com/asrob-uc3m/impresoras-asrob/issues/84#issuecomment-425507793)
> Habilitando el custom menu se añade una opción nueva al menú principal, donde puedes añadir hasta 5 comandos personalizados. Lo ideal es poner ahí lo que más se usa para no tener que estar buscándolo entre los submenús, y además puede haber combinaciones de varias cosas. Todo lo que haya aquí está a 2 clics de la pantalla principal. Por ejemplo se puede configurar así:
> 
> * Precalentar + home.
> * Enfriar.
> * Home.
> * Introducir filamento.
> * Extraer filamento.
> 
> En [506d392](https://github.com/asrob-uc3m/impresoras-asrob/commit/506d392906b08315b79a63e381e4fb5b81fca3f7) y [6c78d1a](https://github.com/asrob-uc3m/impresoras-asrob/commit/6c78d1ae91d9864f3c4328e8c9b3b81d6aa9a9ba) se puede comprobar los comandos que hemos puesto hoy. En un futuro, el uso dirá cuáles es más interesante tener.

Interesante cuanto menos!

Siotma, 2018-10-16:
> Yo probaria el de cambio de filamento, mientras esta en impresion, igual conviene tenerlo mas a mano...

> - Introducir filamento.
> - Extraer filamento.

dmarcosm, 2018-10-17:
> Si pero eso en mitad de la impresion no se si llegaria a funcionar, y creo recordar que habia un comando el nuevo marlin que lo permitia directamente.

Es lo que se llama en Marlin Pausa Avanzada. Básicamente aparca el nozzle a un lado para poder hacer el cambio de filamento. Es interesante pero creo que activar eso y también el autolevel cuando se lo pongamos a geonosis nos va a causar problemas de memoria.

Siotma, 2018-10-19:
Una cosa es actualizar el firmware y otra distinta probar nuevas funciones. Sin duda, los custom commands nos pueden ayudar con la Pausa Avanzada, pero  primero hay que ponerle el nuevo firmware a todas las impresoras y documentarlo bien para poder cerrar esta issue. Luego ya se abrirá otra para jugar con nuevas funciones, sabiendo que siempre podemos volver atrás.

> Añado a actualizar Spicy.

Spicy aún no tiene ficheros de Marlin en el repo :cry:

Siotma, 2018-10-31:
Hoy he flasheado el nuevo Marlin en Hija. Ha habido que hacer varias modificaciones porque había muchas líneas que no estaban marcadas con //Hija_Resurrection. Creo que ya está todo bien. El commit es: 8a2e7166380248934b6436cc1ec6ddb0ca36be21

Siotma, 2018-11-03:
Según tengo entendido, no es trivial actualizar a Spicy y a Geonosis:

- Spicy actualmente lleva un firmware especial para ANET flasheado por @GioSunrider, él puede aportar más datos.
- Geonosis lleva actualmente el firmware original, y no tenemos las fuentes.

Para actualizarlas, se debería abrir otra issue y discutir más detenidamente el modo de proceder.

Creo que esta issue se puede dar por cerrada con éxito y @dmarcosm puede ir haciendo el pull request.

closed 2018-11-03 by Siotma

GioSunrider, 2018-11-04:
Para que conste Spicy esta flasheada con el Marlin denominado SkyNet, en el fondo es un Marlin 1.1.8., actualemente lo tengo disponible en mi perfil de Git, [Link](https://github.com/GioSunrider/X-WING-Anet-A8/tree/master/Marlin%201.1.8%20-%20X-WING). En mi opinion seria conveniente actualizar a Marlin 1.1.9. y fuera.