asrob-uc3m/impresoras-asrob #107: Discusión: renovación de extrusores.

Siotma, 2018-10-19:
Recientemente hemos comprado nuevos extrusores y materiales accesorios: https://github.com/asrob-uc3m/impresoras-asrob/issues/85#issuecomment-431419187

Sin embargo, no está claro qué impresoras van a recibirlos:
- Uno ya está satisfactoriamente instalado en Hija: https://github.com/asrob-uc3m/impresoras-asrob/issues/84
- Quedan dos para repartir entre Blacky, Geonosis, Dédalo y Spicy.

Hay que tener en cuenta que el próximo año con nuevas subvenciones se podrán comprar más.

Esta issue es para decidir en qué impresora se colocarán los extrusores. Luego habrá que abrir otras issues para su colocación, o reutilizar las existentes: #105 #83

Siotma, 2018-10-19:
Pego ideas sacadas de #85 : 

> @Siotma 
> 
> Se han comprado 3 extrusores:
> 
>     * 1 es para reponerme a mi el de Hija.
> 
>     * Los otros dos están disponibles. Habría que ver dónde los colocamos.
>       
>       * Sería interesante ponerle uno a Dédalo por Bowden, ya que mejoraría muchísimo su volumen de impresión.
>       * El otro hay que pensarlo...

> @GioSunrider 
> 
> Por mi parte dejaria a Dedalo en directo por varias razones:
> 
>     * La primera es que me gustaria dejar una impresora en directo, pendientes de lo que pase con Blacky en #83
> 
>     * La segunda es que pensando que hay mucho filamento de 3mm el cambio seria bastante a largo plazo. No me gustaria tener en el armario un extrusor pudiendo mejorar las otras.
> 
>     * La tercera es que pasaria lo mismo con Blacky. Habría que ver si @David-Estevez le parece adecuado. (En el caso de Spicy pasaria lo mismo con @JorFru).
> 
>     * Creo que Spicy mejora una verdadera barbaridad la calidad de impresion colocandole bowden.

> @dmarcosm 
> 
> Yo también soy partidario de ponérselo a Dedalo. Primero, tendríamos el mismo diámetro de filamento en todas las impresoras. No hay una necesidad extrema de usar los filamentos de 3mm que tenemos, porque están todos bastante "pasados" digamos. Y segundo por lo que comenta @Siotma del espacio de impresión. Ahora mismo no llega a los 100mm en Z...
> El segundo se lo pondría a la Anet pero dejándolo directo.

Siotma, 2018-10-19:
Mi primera aportación nueva es que, siendo los extrusores de ASROB, y siendo especiales para bowden, sería una pena que Geonosis se quedara sin uno (Geonosis e Hija son las únicas impresoras propiedad de ASROB y debería ser una prioridad que funcionen adecuadamente).

Por otra parte, sigo pensando que Dédalo es un buen candidato a obtener uno de estos extrusores teniendo en cuenta que tiene el extrusor más antiguo de todos, y que además condiciona mucho su volumen de impresión.

A Blacky: le vendría bien un extrusor como los que hemos comprado porque ahora se atasca de vez en cuando, pero hay que tener cuidado con cambiarle piezas porque es del departamento.

A Spicy: es una impresora nueva y funciona correctamente, por lo que no veo urgencia en cambiar su extrusor.

dmarcosm, 2018-10-19:
Yo a Geonosis no le cambiaría el extrusor, el que trae me parece que es decente (al menos más decente que el de Spicy). 
Mantengo lo de que se lo pondría a Spicy pero sería conveniente tener alguna impresora con extrusor directo.

Siotma, 2018-10-19:
Hay que tener en cuenta que para instalar un extrusor como el puesto en Hija (#84) hay que tener los siguientes elementos:
- [1x Hobbed Bolt](https://fervi3d.com/extrusion/156-tornillo-ranurado-extrusor-m8.html)
- [3x Rodamiento 608ZZ](https://fervi3d.com/mecanica/154-rodamiento-608zz-8x22x7mm.html)
- 1x Eje 8mm diametro 20 mm largo

En su momento yo doné un kit para Hija en https://github.com/asrob-uc3m/impresoras-asrob/issues/42#issuecomment-362639453 , pero lo suyo no es eso. Actualmente sólo Hija, Blacky y Dédalo tienen ya estas piezas. Para el resto habría que comprarlas.

GioSunrider, 2018-10-20:
> Mi primera aportación nueva es que, siendo los extrusores de ASROB, y siendo especiales para bowden, sería una pena que Geonosis se quedara sin uno (Geonosis e Hija son las únicas impresoras propiedad de ASROB y debería ser una prioridad que funcionen adecuadamente).
> 
Esta noche pense lo mismo y por las mismas razones, ademas de que poniendole un reductora tendriamos la posibilidad de que no perdiera pasos al extruir.

> Hay que tener en cuenta que para instalar un extrusor como el puesto en Hija (#84) hay que tener los siguientes elementos:
> 
> * [1x Hobbed Bolt](https://fervi3d.com/extrusion/156-tornillo-ranurado-extrusor-m8.html)
> * [3x Rodamiento 608ZZ](https://fervi3d.com/mecanica/154-rodamiento-608zz-8x22x7mm.html)
> * 1x Eje 8mm diametro 20 mm largo
> 
> En su momento yo doné un kit para Hija en [#42 (comment)](https://github.com/asrob-uc3m/impresoras-asrob/issues/42#issuecomment-362639453) , pero lo suyo no es eso. Actualmente sólo Hija, Blacky y Dédalo tienen ya estas piezas. Para el resto habría que comprarlas.

Referido a esto añadiria a la subvencion uno o dos kits de estos, ya que podemos prescindir de un filamento y comprar esto que casi es mas necesario ademas de tener asi respuestos.

dmarcosm, 2018-10-20:
> Yo a Geonosis no le cambiaría el extrusor, el que trae me parece que es decente (al menos más decente que el de Spicy).

Con esto quería decir que hay que pensar que no es lo mismo extrusor que hotend. Pienso que Geonosis tiene un hotend aceptable, y que lo que necesita ser cambiado es el extrusor, por esto: 
> poniendole un reductora tendriamos la posibilidad de que no perdiera pasos al extruir.

GioSunrider, 2018-10-21:
Yo a Geonosis le cambiaba todo, nos quitaríamos de muchos problemas. Y lo que dice Manu, son la prioridad que funcionen bien.

dmarcosm, 2018-10-21:
Pero si ese hotend ya funciona bien, no es necesario cambiarlo "para que funcione bien", porque ya lo hace. Cambiando el extrusor ya nos quitaríamos de "todos" los problemas y dejaríamos el otro hotend libre para otra impresora.

Siotma, 2018-10-23:
Propongo mañana desmontar el hotend de Geonosis y valorar opciones de mejorarlo.

GioSunrider, 2018-10-23:
Me parece perfecto.

Siotma, 2018-10-27:
Además, téngase en cuenta esto respecto a Spicy: https://github.com/asrob-uc3m/impresoras-asrob/issues/105#issuecomment-433620314

GioSunrider, 2018-11-27:
La propuesta es, Colocar un volcano a Geonosis con el V6 dado a su volumen de impresion sea mas aprovechado, ademas de a Blacky, y el Hot-End de Geonosis colocarselo a Dedalo cuando se gaste todo el filamento de 3 mm que actualmente tenemos.