asrob-uc3m/impresoras-asrob #109: Engranajes para la OSHWDem 2018

GioSunrider, 2018-10-24:
Tal y como se comento por el grupo de Telegram, hay un reto de "Clone Wars" para la OSHWDem 2018. Las especificaciones del reto se encuentran en [este repositorio](https://github.com/brico-labs/Reto-Clonewars-Oshwdem-2018). 
El reto consiste en llenar una pared de engranajes, y que mejor que nosotros para diseñar los que llevaremos como asociacion. Mi idea es llevar un minimo de 2, uno como asrob y otro como Printe One.
El primero de asrob seria el N34: [Reto Clone Wars OSHWDem 2018](https://github.com/asrob-uc3m/impresoras-asrob/tree/master/Reto%20Clone%20Wars%20OSHWDem%202018)

GioSunrider, 2018-10-24:
Tambien habria que diseñar uno que ponga Printer One e imprimirlos. ¿Algun voluntario?

dmarcosm, 2018-10-24:
> El primero de asrob seria el N34: Reto Clone Wars OSHWDem 2018

Sube el stl, porfi

GioSunrider, 2018-10-24:
No consigo generarlo, puedes intentarlo tu?

GioSunrider, 2018-10-25:
Al generar el gcode daba error de pieza, se ha pasado por una aplicación web que @Siotma recomendó y se han reparado los STL. Actualmente los archivos correctos son las extensiones .3mf que vienen indicados como repaired.

GioSunrider, 2018-10-25:
Además se vio que la impresión era demasiado larga, por lo que se propone como reto de impresión en casa, ¿algún voluntario?

GioSunrider, 2018-10-26:
Lo he probado, y no termina de generar bien el STL Como con esta pagina.

dmarcosm, 2018-10-26:
Pues a mí me lo acaba de generar sin problemas, habrá que actualizar el Slic3r del PC de la nave por si fuera cosa de la versión. Lo digo sobre todo de cara a no tener que andar con ñapas, que es lo que es abrir una página más

GioSunrider, 2018-10-26:
Tenemos al rey de las ñapas en nuestras filas, no creo que las papas sean el problema 😂😂😂😂

David-Estevez, 2018-10-26:
🤟🏻

GioSunrider, 2018-10-27:
He puesto a imprimir en Negro el Engranaje de Printer One. Dejo pendiente de quien lo desee el de Asrob, a ser posible que no sea ni blanco ni negro.

Siotma, 2018-10-28:
La página de la que habla @GioSunrider es https://tools3d.azurewebsites.net/ (editado también en su comentario). Es de uso libre, solo hace falta una cuenta de microsoft (el hotmail del messenger mismamente).

> Como dato, y para evitar tener que usar otros sitios, Slic3r auto repara el STL cuando ve que tiene errores. Al importarlo lo hace automágicamente :)

No conozco ninguna herramienta de reparación capaz de hacer lo que hace esta.

Siotma, 2018-10-28:
Leyendo las [condiciones del reto en la página oficial](https://github.com/brico-labs/Reto-Clonewars-Oshwdem-2018), veo que los archivos disponibles para el engranaje N-34_ASROB **no cumple con los requisitos**, ya que no tiene agujero pasante. Además, tampoco veo que nos hayamos apuntado en la [lista de participantes](https://docs.google.com/spreadsheets/d/1SnokEGJ45Vu7sOCrKUwsjmvuEwmsKIhkQNeIIfKUH_w/edit#gid=0)...

@GioSunrider, alguna idea?

GioSunrider, 2018-10-28:
Tenia pensado apuntarnos, una vez impresos correctamente, por si no daba tiempo a imprimirlos.

GioSunrider, 2018-11-08:
Hija imprimió el ultimo engranaje.

[image]

GioSunrider, 2018-11-08:
Engranajes guardados y listos para llevar a la OSHWDem!!

GioSunrider, 2018-11-12:
Se entregaron los engranajes y quedaron colocados en la pared! ademas adjunto las fotos y videos en los que se pueden ver lo bien que han quedado.

[Publicacion Instagram ASROB](https://www.instagram.com/p/BqA1MprHL6A/?utm_source=ig_web_options_share_sheet)

[image]

Con esto doy por concluido el reto y cierro issue.

closed 2018-11-12 by GioSunrider