asrob-uc3m/impresoras-asrob #112: Hacer ballestas

VentressAsajj, 2018-11-12:
Me acordé de ponerlo @GioSunrider @dmarcosm ¿cuándo hacemos las ballestas? 
Tweet:
https://twitter.com/fisadev/status/1059631656250806274
Thingiverse:
https://www.thingiverse.com/thing:1995091
Cómo montarla
https://t.co/8zYOswTtnr

GioSunrider, 2018-11-12:
Organizamos un taller de ballestas 😂😂

VentressAsajj, 2018-11-13:
Vamos. Cuando es la próxima reunión de printerone? yo tengo que ser operadora, debo poner otra issue ( como digas que sí te mato jajaja )

MrTommyGuns, 2018-11-13:
La próxima reunión es Jueves de 6 a 9

GioSunrider, 2018-11-13:
Eso es sigue lo que te ha dicho Tommy, vivan las issues 🤣🤣

dmarcosm, 2018-11-13:
> Vamos. Cuando es la próxima reunión de printerone? yo tengo que ser operadora, debo poner otra issue ( como digas que sí te mato jajaja )

El jueves las imprimimos!!

VentressAsajj, 2018-11-13:
Pues si el jueves se imprimen, no creo que sea necesario hacer el curso intensivo de operadora ni poner la puuuññññetera issue :joy:

dmarcosm, 2018-11-13:
Imprimir las piezas de la ballesta te vale como parte de la formación de operadora :)

jorgercobo, 2018-11-15:
Primer intento de impresión

jorgercobo, 2018-11-15:
Falta una de las piezas laterales y los tornillos

closed 2023-09-28 by CarlosME17