asrob-uc3m/impresoras-asrob #114: Cambio extrusor para Geonosis

jorgercobo, 2018-11-15:
Se ha decidido cambiar el extrusor de Geonosis a un v6 con bloque Volcano de nozzle 1,2. Para ello se necesitan las siguientes piezas:

- 3 piezas del [carro](https://www.thingiverse.com/thing:2865870):
  - [ ] [1x Duct.stl.zip](https://github.com/asrob-uc3m/impresoras-asrob/files/2586589/Duct.stl.zip)
  - [ ] [1x CR-10_HD-modular-mount-V6_mount-cover__40mm-fan_v1.stl.zip](https://github.com/asrob-uc3m/impresoras-asrob/files/2586593/CR-10_HD-modular-mount-V6_mount-cover__40mm-fan_v1.stl.zip)
  - [ ] [1x v6_mont_x5s_v4.stl.zip](https://github.com/asrob-uc3m/impresoras-asrob/files/2586594/v6_mont_x5s_v4.stl.zip)
  - [ ] [(Opcional) 1x autolevel.stl.zip](https://github.com/asrob-uc3m/impresoras-asrob/files/2586587/autolevel.stl.zip)

GioSunrider, 2018-11-27:
Imprimiendolo en casa para ver si esta tarde lo puedo llevar para comenzar su montaje.

Siotma, 2018-11-27:
Imprimiendo el duct en Hija.

jorgercobo, 2018-11-27:
Se ha comenzado a montar el carro, pero hay un problema con las tuercas traseras. Pendiente de volver a montarlo en la siguiente reunión

jorgercobo, 2018-11-27:
[image]

GioSunrider, 2018-12-07:
Se imprimio en Blacky el soporte del Autolevel, ya que, como se ve en la foto, es donde tambien va colocado el End-Stop. 

[image]

Ademas como se ven en la foto anterior se termino de colocar el resto del extrusor a falta de colocar el ventilador de capa, el cual se realizo el cambio y se soldó. 

[image]

No se pudo colocar, ya que el Duct no esta adaptado para la cabeza Volcano, para ello, he diseñado una [pieza nueva para poder poder el ventilador](https://www.thingiverse.com/thing:3265261)
El proximo dia se imprimira y se colocara.

GioSunrider, 2018-12-20:
En la pasada reunion lleve el nuevo duct, esta vez, con los cambios necesarios. Se instalo y quedo perfecto. por lo que queda como satisfactorio el cambio de extrusor.

[image]

closed 2018-12-20 by GioSunrider

JorFru, 2019-05-09:
El nozzle actualmente instalado en geonosis, aunque en su inscripción lateral se puede leer claramente 1.2, este nozzle es de 0.8mm aunque Giorgio diga lo contrario.
El calibre digital chino dicta que son 0.8, y los chorretes que suelta miden 0.8 también.
conclusión: 0.8