asrob-uc3m/impresoras-asrob #115: Tensar y alinear la correa del eje Y de Dédalo

Siotma, 2018-11-30:
La correa del eje Y de Dédalo quedó destensada tras finalizar #69 . Hay que tensarla, para lo cual hay que desmontar la cama caliente. Propongo que se aproveche la ocasión para alinear la correa ya que actualmente tiene cierto ángulo y esto provoca esfuerzos e imprecisiones. Propongo imprimir una pieza sencilla (un prisma sería suficiente) que haga de suplemento.

CeciliaVa, 2018-12-05:
Hoy @MrTommyGuns, @CinnabarCinnamon y yo hemos alineado y tensado la correa, aunque no hemos impreso ningún tipo de prisma supletorio. No sé si esta pieza era necesaria o era sólo una sugerencia.

GioSunrider, 2018-12-15:
Revise la alineacion de la correa y me pareció correcta. Yo tampoco se a que prisma se refiere @Siotma pero la impresora en teoría debería funcionar sin problemas.

GioSunrider, 2018-12-15:
@CeciliaVa o @MrTommyGuns proceder alguno a cerrar la issue que os lo habéis ganado.

MrTommyGuns, 2018-12-15:
Cerraamos!

closed 2018-12-15 by MrTommyGuns

Siotma, 2019-01-11:
Básicamente me refería a instalar algún suplemento bajo la cama que haga que el movimiento de la correa sea horizontal y no inclinado, de esta forma el polígono que circunscribe la correa no se deforma al moverse la cama, lo que evita tensiones y destensiones indeseadas.

Me he entretenido dibujando esta imagen para que se entienda lo que quiero decir: como se puede ver, entre la posición extrema de la cama (1mm) y la posición central de la cama (90mm) la forma de la correa (línea verde) cambia si esta está desnivelada; pero estando nivelada siempre mantiene la misma forma:
[image]

Si echamos cuentas sacamos que la correa cuando está desnivelada puede sufrir esfuerzos de tensión y destensión, llegando a modificar su longitud significativamente, mientras que estando nivelada eso no ocurre:
[image]

A la vista de esto, considero que tener la correa nivelada alarga la vida de la mecánica y también mejorará la precisión de las impresiones.

Siotma, 2019-01-11:
Por lo descrito en el comentario anterior, ayer añadí unos suplementos bajo la cama de dédalo que corregían el desnivel:

[image]

Siotma, 2019-01-12:
Las piezas, por cierto, las saqué de aquí: https://www.thingiverse.com/thing:181524/files

[belt-y-clamp-gt2.zip](https://github.com/asrob-uc3m/impresoras-asrob/files/2751964/belt-y-clamp-gt2.zip)