asrob-uc3m/impresoras-asrob #123: Tensores Spicy X e Y

GioSunrider, 2019-01-13:
En la ultima reunion, aprovechando que #122 vino con el Tensor del eje Y, se desmonto e instalo en spicy, dando como resultado la siguiente imagen.

[image]

El siguiente paso a recorrer seria la discusion, impresion e instalacion del tensor del eje X, por lo que propongo el siguiente proceso:

- [ ] Discusión sobre que tensor colocar

- [ ] Impresion del Tensor

- [ ] Colocacion del Tensor

CinnabarCinnamon, 2019-02-07:
Piezas encontradas para tensor:

[X-axis-Right_2017.zip](https://github.com/asrob-uc3m/impresoras-asrob/files/2842201/X-axis-Right_2017.zip)
[BIG-thumbscrew.zip](https://github.com/asrob-uc3m/impresoras-asrob/files/2842202/BIG-thumbscrew.zip)
[pulley-Toothed_bearing.zip](https://github.com/asrob-uc3m/impresoras-asrob/files/2842203/pulley-Toothed_bearing.zip)
[tensioner-Thick.zip](https://github.com/asrob-uc3m/impresoras-asrob/files/2842204/tensioner-Thick.zip)

GioSunrider, 2019-02-14:
ya que hay que imprimirm, imprimiremos mejor la misma pieza que lleva [Hija](https://github.com/asrob-uc3m/impresoras-asrob/tree/master/Hija%20Resurrection/Parts/X%20axis)

CinnabarCinnamon, 2019-02-14:
La pieza impresa sirve para Blacky, NO para Spicy.

GioSunrider, 2019-02-15:
Sigue para Blacky en #136

GioSunrider, 2019-02-15:
Para no tener que imprimir una pieza nueva del eje X, propongo el siguiente [tensor](https://www.thingiverse.com/thing:3078070) asi ni necesita sustitucion de correa por ser demasiado pequeña ni una piezaq entera nueva. El proximo dia, podemos imprimir y colocarlo.

MrTommyGuns, 2019-02-21:
Adjunto Link para imprimir un tensor bien hot.
https://www.thingiverse.com/thing:1428253

L4g0, 2019-02-28:
Instalado tensor eje X :smile: 
[image]

closed 2019-02-28 by L4g0