asrob-uc3m/impresoras-asrob #127: Video 1. Presentacion de la granja

GioSunrider, 2019-02-07:
En esta issue reflejaremos el guion y la produccion del video de presentacion de la granja, asi como el material audiovisual del mismo.

GioSunrider, 2019-02-07:
Imagenes de las impresoras para utilizar.

https://drive.google.com/file/d/1Esr2Bw1mX4VhRgQrtKhgr3Jn0qyi5ILj/view?usp=sharing

L4g0, 2019-02-07:
## Guión 

¡Hola! Bienvenidos al curso de manejo e impresión 3D de asrob uc3m.

Para empezar os vamos a presentar a nuestras máquinas.

Del proyecto Clone Wars han nacido **Hija Resurrection**, **Blacky** y **Dédalo**

La primera es una prusa i3 steel (más info [en la descripición](http://wiki.asrob.uc3m.es/index.php/RepRap:_HIJA)). Actualmente se encuentra con un nozzle de cosas 

La segunda se trata de una Prusa Air 3 (más info [en la descripición](http://wiki.asrob.uc3m.es/index.php/RepRap:_BLACKY))

**Dédalo** se trata de una Prusa i2. Esta es la única de nuestras impresoras que utiliza filamento de 3 mm (más info [en la descripición](https://www.reprap.org/wiki/File:Clon-28-Dedalo.jpg))

**Geonosis** es la última de nuestras adquisiciones. Es un modelo TronXY X5S y es la única impresora CoreXY que tenemos por el momento. 

**Spicy** (caldito para los amigos) es el conocido modelo Anet A8, adquirida a través de una donación.

closed 2023-09-28 by CarlosME17