asrob-uc3m/impresoras-asrob #135: Charla Impresoras 3D en Jornadas Técnicas del GUL

JorFru, 2019-02-14:
Hola, abro issue para pedir préstamo.

Me he presentado para dar una charla en las JT del GUL https://cursos.gul.es/xxxiii-jornadas-tecnicas-del-gul

Me gustaría llevar a la charla a Dédalo, para que los asistentes vean las abominaciones mecánicas que se construían con pocos recursos, a fin de calmar el SAV.
El objetivo es llevarla para enseñar, quizás ponerla a imprimir pero sería raro.

Más datos: jueves 18 febrero, de 16 a 18h, 1.1.G04.
Si alguien se anima a ayudarme con la charla, invitado está.

GioSunrider, 2019-02-14:
Ahora mismo dedalo no puede imprimir a falta de terminar su cable managment. Por el prestamos sin problema, yo mismo puedo echarte un cable con la charla si lo necesitas.

JorFru, 2019-02-14:
Okey, gracias por la info y el ofrecimiento.
Continuamos la colaboración por chat privado.
Si alguien más se anima que se manifieste

David-Estevez, 2019-02-15:
>  las abominaciones mecánicas que se construían con pocos recursos, a fin de calmar el SAV.

:unamused:

GioSunrider, 2019-02-26:
Dejo aqui la foto del evento con dedalo habiendo acabado de imprimir y @JorFru exponiendo de fondo.

[image]

JorFru, 2019-05-08:
Aquí dejo la documentación generada
https://github.com/JorFru/Charlas/tree/master/JornadasT%C3%A9cnicasGUL/XXXIII

Desde aquí otra vez agradecer a Coleóptero, Obijuan y todo Clone Wars.
Sobre todo por la cantidad de diapositivas que dejaron de sus charlas anteriores y he podido reutilizar.

closed 2019-05-08 by JorFru