asrob-uc3m/impresoras-asrob #137: Mejoras Spicy

MrTommyGuns, 2019-02-21:
Abro esta issue para ir documentando y mejorando a Spicy.

Propongo esta página para ir mirando las mejoras: https://www.staticboards.es/blog/mejoras-anet-a8/

- [ ] Botón para encendido. _(Necesaria aprobación @JorFru )_

- [ ] Mejora estructura eje X.

- [ ] Tensor eje X. _(Issue #123 )_

- [ ] Final de carrera Z.

- [ ] Aros para muelles cama caliente.

- [ ] Reforzar interior delantero.

- [ ] Reforzar interior trasero.

- [ ] Soldar cables directamente a la cama. _(Necesaria aprobación @JorFru )_

GioSunrider, 2019-11-15:
En la reunion de ayer se imprimi e instalo la mejora del final de carrera del Eje Z.

MrTommyGuns, 2019-12-06:
Si no recuerdo mal, en la reunión de ayer @JorFru y yo comentamos la posibilidad de ponerle un extrusor nuevo a Spicy y chetarla un poquete ya que tenemos extrusores para cambiar en el armario.

Esto podría ser una mejora bastante guay para Spicy.

GioSunrider, 2019-12-07:
A mí modo de ver prefiero que el extrusor bondtech del armario vaya para blacky o geonosis, y me decanto más por geonosis, ya que el motor sufre mucho y ponerle la reductoras ayudara a este problema

closed 2023-09-28 by CarlosME17