asrob-uc3m/impresoras-asrob #139: Nos donan una Cube Pro

JorFru, 2019-03-01:
Hola
Pues eso, que nos regalan una impresora casta.

Un profesor de bioingeniería nos la regala porque se han comprado una MK3 y les funciona mejor.
Nos sugiere que le hagamos una conversión al lado del bien (open source).

Le he sugerido quedar el 4 o 5 de marzo de 2019, para ir a verla y traerla si nos mola. Voy yo. Si alguien quiere acompañarme, quedamos. Cuando me conteste pongo fecha y hora.

Hay que empezar a pensar un nombre para ella jijijiji

GioSunrider, 2019-03-02:
Ueeee palantisimo, tenemos además material de sobra para dedicarle.

JorFru, 2019-03-05:
Lista de modificaciones (por orden de urgenicia)

- [ ] Trucar el tema cartuchos de filamento. (cuesta 100€ cada 900€ de filamento oficial)

- [ ] Instalar electrónica libre. (Da mucho morbo y es necesario para trastear)

- [ ] Instalar sistema multifilamento (multicolor y/o multimaterial).

Por parte del antiguo dueño estamos completamente autorizados a todo esto.

MrTommyGuns, 2019-10-10:
Esto deberíamos seguirlo en #142 (?)

GioSunrider, 2019-10-20:
Afirmativo, continuamos en #142

closed 2019-10-20 by GioSunrider