asrob-uc3m/impresoras-asrob #140: PROPUESTA: Donación Plotter_LSM

Lucas-LSM, 2019-03-21:
Hola gente de ASROB:

Tengo un plotter freehardsoftware, (CNC arduino) para donar. Os dejo link de youtube de cómo es:

https://www.youtube.com/watch?v=-GkaaUIY3kc

Es una copia del plotter de 4xiDraw

Qué decís?

GioSunrider, 2019-03-21:
No se como andamos de espacio, pero por mi adelante, cuanto mas mejor xD

JorFru, 2019-04-03:
Actualizo con que tenemos muy poco espacio disponible.
Necesitaríamos upgradear a una segunda estantería.
De hecho la CNC de @DiegoGarciaX2 la tenemos ahora mismo en un emplazamiento poco adecuado, y también habría que buscarla hueco.

¿Qué os parece mover las cajas de filamento de la balda baja de la estantería a la balda última?
Así dejaríamos la CNC de Diego abajo, y tendríamos hueco para el plotter de Lucas.
Las cajas cerradas no cogen polvo, y las nuevas máquinas también quedarán protegidas del polvo.

MrTommyGuns, 2019-04-03:
Me parece una buena solución lo de pasar las de filamento arriba y aprovechar la balda baja, al menos por el momento.
Si podemos en el futuro tener 2 estanterias o algo del estilo ya se verá como organizarlo mejor.

JorFru, 2019-04-03:
Calculo que, de conseguir estantería por la vía habitual, llegaría mínimo en marzo del 2020.

GioSunrider, 2019-04-04:
Por mí sin problema también.

Lucas-LSM, 2019-04-04:
Pues cuando lo tengais y me digais la acerco

Obtener Outlook para Android<https://aka.ms/ghei36>

JorFru, 2019-05-08:
Todavía estamos organizando el espacio y no tenemos mucho disponible. Próximamente te avisamos.
Gracias Lucas.

closed 2023-09-28 by CarlosME17