asrob-uc3m/impresoras-asrob #141: Soporte de filamento para Spicy

MrTommyGuns, 2019-05-08:
Hace unos meses el soporte para filamento de Spicy murió.
El tema está en que tenemos varias impresoras que imprimen correctamente y nos faltan soportes si las queremos poner a imprimir todas.
Sugiero la siguiente pieza para hacerle un soporte:

https://www.thingiverse.com/thing:2162266

Por el momento la he impreso para mi anet y parece que funciona correctamente.

MrTommyGuns, 2019-05-08:
Si lo hacemos, sugiero ponerle una capa de pegamento caliente fina como antideslizante y la estabilidad mejora una barbaridad

GioSunrider, 2019-05-09:
Buena idea, se esta imprimiendo en Hija ahora mismo.

MrTommyGuns, 2019-05-09:
En la tarjeta de hija está el gcode del soporte y los conos están en la de Spicy, por si queréis reutilizarlos...
A los conos les faltaba poner algo más de relleno

GioSunrider, 2019-05-09:
he borrado la tarjeta entera... eso no se deja ahi...

GioSunrider, 2019-05-09:
Se han impreso las dos piezas, una vez se coloquen se subira la foto y quedara por cerrada la issue.

JorFru, 2019-05-09:
[image]

JorFru, 2019-05-09:
Mejoras de pegamento caliente por @MrTommyGuns

closed 2019-05-09 by JorFru