asrob-uc3m/impresoras-asrob #142: Misión: Descastarización

MrTommyGuns, 2019-05-17:
Ayer comenzamos el proceso para democratizar la cube.

Realizamos lo siguiente:

- Desmontar el "suelo".

- Sacar la placa controladora.

- Desenchufar todos los cables y mirar la potencia de la fuente (24V).

- Investigar la electrónica, cables y clasificar mediante etiquetas.

JorFru, 2019-05-17:
Yo fui quien cortó los tres cables del termómetro.
Si alguien me quiere culpar, disponible estoy.

Es que está muy mal diseñado macho, tiene cables que llegan por un tubo y que la mayoría de esos cables van a un conector pero algúno otro va a otro conector de otra parte de la placa. Y se meten destrangis por sitios raros.

El asunto es que lo corté porque seguramente no vayamos a usar esa sonda de temperatura por el momento, el cable estaba dando mucho por culo para el desmontaje. Y cuando lo usemos ya tendremos tiempo de conectorizar ese cable o tomar decisiones. Además primero debemos de caracterizar el sensor y ver si está en Marlin.

GioSunrider, 2019-10-20:
Seguido de #139 
> Lista de modificaciones (por orden de urgenicia)
> 
> * [ ]  Trucar el tema cartuchos de filamento. (cuesta 100€ cada 900g de filamento oficial)
> * [ ]  Instalar electrónica libre. (Da mucho morbo y es necesario para trastear)
> * [ ]  Instalar sistema multifilamento (multicolor y/o multimaterial).
> 
> Por parte del antiguo dueño estamos completamente autorizados a todo esto.

MrTommyGuns, 2019-12-06:
Ayer se abordó el tema otra vez con muchas ganas hasta que vimos que necesitamos electronica a 24V que no tenemos (la que tenemos funciona a 12V).

Una de las opciones propuestas es adquirir una electrónica de Ender 3, que funciona a 24V.

JorFru, 2019-12-08:
https://dyzedesign.com/2018/09/cube-pro-total-conversion-step-by-step-guide/
Documentación del proceso de descastización por parte de otra gente que ha hecho lo mismo con la bicha.

MrTommyGuns, 2019-12-08:
Muy buena documentación!

Eso sí, la lista de materiales se va un poco de precio jajaja

GioSunrider, 2019-12-08:
Se podria meter como subveccion, denmocratizarla asi que no veo problema en el precio.