asrob-uc3m/impresoras-asrob #143: Conos para filamento

MrTommyGuns, 2019-05-22:
Creo muy conveniente que imprimamos una serie de conos, tubos o lo que sea para centrar los filamentos en los soportes.

En mi opinion son más útiles los conos porque se adapta a los tropecientos tipos de bobinas que tenemos, pero es más engorroso colocarlos si no se plantea el mecanismo adecuadamente.

Ejemplo: [https://www.thingiverse.com/thing:767317](url)

fervh, 2019-10-24:
[cono.zip](https://github.com/asrob-uc3m/impresoras-asrob/files/3768816/cono.zip)

MrTommyGuns, 2019-11-26:
Finalmente se realizó la impresión de 2 de estos [conos](https://www.thingiverse.com/thing:767317)
(Adjuntaré foto cuando sea posible)
Creo que al menos deberíamos hacer un par más.