asrob-uc3m/impresoras-asrob #144: Creación de Perfiles de Impresión

MrTommyGuns, 2019-05-22:
Creo que deberíamos tener una serie de perfiles básicos creados para cada impresora, para que cada persona que quiera utilizar las impresoras no se tenga que liar a hacer un perfil que muy probablemente de errores.

Los que hay subidos creo que están muy desactualizados ya que las maquinas han cambiado bastante en algunos casos.

Por lo que he visto la mayor parte de la gente que usa las impresoras usa Cura, pero por otro lado nunca está demás tener perfiles de diferentes slicers.

Propongo que a medida que se van utilizando las impresoras, si el resultado de impresión ha sido aceptable, se suba el perfil usado para partir de una base utilizable que podamos usar habitualmente y fijándonos en los detalles ir puliendo el perfil. Tal vez incluso si usaramos todos el mismo perfil podríamos aportar mejoras consiguiendo un punto de impresión óptimo, pero siempre hay detalles que gustan a unos más que otros así que siempre se pueden retocar a gusto del consumidor.

Resumiendo, tener unos buenos perfiles de base sería muy necesario en mi opinión.

GioSunrider, 2019-05-22:
Me parece muy buena idea, así podemos ir perfilando distintos tipos de slicers y los vamos puliendo.

MrTommyGuns, 2019-12-06:
Actualmente, tenemos los perfiles de Hija, Spicy y Blacky, que imprimen bastante bien partiendo del mismo perfil aunque molaria personalizarlos para cada una, mejorandolo para las prestaciones de cada una.

Por otro lado, se ha creado un perfil de impresión que fue un buen punto de inicio para Dedalo por parte de @DiegoGarciaX2 y otro para la nueva impresora por parte de @MarioPrinter.

Todos ellos han sido creados para Cura y se irán subiendo a la carpeta correspondiente para cada una.

DiegoGarciaX2, 2019-12-07:
Donde subo el perfil?

DiegoGarciaX2, 2019-12-07:
el la carpeta de dedalo?

GioSunrider, 2019-12-08:
Sigue la estructura de las carpetas y subirlo cada uno donde corresponde. Y a ser posible mencionar el commit en la issue para tenerlo referenciado.

closed 2023-09-28 by CarlosME17