asrob-uc3m/impresoras-asrob #145: Electrónica de Geonosis

JorFru, 2019-06-04:
Abro issue nueva para hablar de la placa nueva. La conversación anterior está en #95
Después de la vuelta a la electrónica original de Geonosis, me encuentro con que no funciona, la semana pasada sí funcionaba bien.

¿Qué fue lo último que se hizo?
¿Se han hecho pruebas de impresión después de atornillar la placa?
P.D.: issue de cable managing pronto

MrTommyGuns, 2019-06-04:
Lo último que se hicieron fueron pruebas de impresión...
Mientras se hacía una de ellas estabamos montando el ventilador y se paró porque presuntamente le di al botón se reset y no volvimos a ponerla a imprimir porque era tarde y así poder realizar el montaje más cómodamente...
El otro día yo intenté usarla y la pantalla no funcionaba...
Se lo comenté a Gio y me dijo que estaba dando problemas pero que sí que iba.
Eso es toda la info que tengo

JorFru, 2019-06-04:
a mí tampoco me ha funcionado

Añado estatus:
Pantalla en blanco (bueno, azul)
Por USB reconoce el CH340 pero no el puerto serie.

Pasos que sugiero:
Desmontar la placa de todos los tornillos y probarla al aire. Parece que algo ha salido mal en el montaje.
Puede ser que se esté reiniciando todo el rato? No sé.

GioSunrider, 2019-06-05:
Primera labor para hacer mañana!

JorFru, 2019-07-02:
La pasada semana le eché un vistazo y tiene mucha pinta de electrónica frita.
Ahora que tenemos RAMPS y MEGAs, podríamos hacer un cambio de la electrónica de Geonosis.

GioSunrider, 2019-07-19:
Ayer @MrTommyGuns y yo, recableamos la impresora, y se instalo una OVM 20 Lite. Se ha subido el Marlin en a37f707. Queda pendiente hacer Octoprint ya que no tenemos adaptador de pantalla y la hemos perdido, ademas de la impresion por SD.

closed 2019-07-19 by GioSunrider