asrob-uc3m/impresoras-asrob #146: Limitaciones Imaginarias

MrTommyGuns, 2019-06-05:
No sé como se llamará esto, pero resulta que Hija y Blacky (si no recuerdo mal), tienen un error de "bloqueo" en la cama por lo que sea, provocando que cuando se va a imprimir no se pueda usar todo el espacio de la cama.

JorFru, 2019-06-05:
Hija y Blacky tienen una cama más grande de la configurada en Marlin
Por eso "sienten" una limitación del movimiento de la cama que trunca las piezas antes de llegar al borde de la cama.
También hay que comprobar el X e Y offset a la hora de slicear.
Normalmente Cura trae un par de centímetros muertos en X e Y. Esta es la distancia entre el punto de homing y el principio de la cama. No en todas las impresoras es el mismo.
No sé si este parámetro también está en Marlin. De ser así se juntan los dos y termina imprimiendo en la esquina posterior derecha.

MrTommyGuns, 2019-12-06:
Actualmente hemos realizado pruebas de impresión y las impresoras imprimen con buena calidad. No se han observado estos fallos por lo que cierro la issue hasta que vuelvan a observarse.

closed 2019-12-06 by MrTommyGuns