asrob-uc3m/impresoras-asrob #150: Uso de filamento por los operadores

MrTommyGuns, 2019-10-13:
Deberíamos resolver cuanto antes el problema/dilema de la cantidad de plástico de la asociación usado por los operadores.

La mejor opción yo diría que es poner limites en los gramos de las impresiones, de tal modo que si se utiliza este plástico no sea de forma abusiva para un uso personal.

A parte de eso habría que poner un límite en el cual se dejará de poder imprimir con el plástico de forma personal (punto para dar una palmadita en el hombro y decirle al susodicho que se gaste 10€ y compre una bobina xD).

GioSunrider, 2019-10-13:
Esto creo que iría mejor en actas. Y de hecho creo que ya hay issue, aunque esté cerrada.

jgvictores, 2019-10-14:
> Esto creo que iría mejor en actas. Y de hecho creo que ya hay issue, aunque esté cerrada.

Síp, era la intención de https://github.com/asrob-uc3m/actas/issues/61. Aunque se trate de lo mismo, me gusta más cómo se plantea el problema aquí. Si lo queréis mover, ahí está el botón de `Transfer issue`. :smiley:

MrTommyGuns, 2019-11-26:
Teniendo en cuenta como se está llevando finalmente este asunto, creo que hemos dado con una buena dinámica.
A todos los nuevos operadores se les ha dicho que para imprimir lleven su propio filamento en el caso que sea para uso personal.
Si es para uso interno de la asociación, ya sean mejoras para las impresoras, proyectos de otras ramas... Se usará el filamento de la asociación.
Como siempre, puede que haya situaciones en las que se pueda usar el filamento de la asociación para uso personal, siempre que sea de manera puntual, haya algún punto de interés y/o se compense por otro lado (casos más concretos, que a mi parecer tampoco pasa nada).
Pero dejaría cerrado el asunto con esta propuesta si os parece bien.

GioSunrider, 2019-11-27:
A mi me parece lo correcto.

David-Estevez, 2019-11-27:
[image]

jgvictores, 2019-11-27:
Opino lo mismo. Parece buena idea, pero tb parece una cosa muy horizontal.

Mejor una PR sobre estatutos, y votar sobre el merge.

Como pequeño aporte: ¿Quizás alguna cláusula de pago comisión en fungible? No sé, si alguien va a pulirse 5-10 bobinas, pues tal vez tenga sentido que aporte 1 bobina a la Asociación por compensar desgaste o algo.

GioSunrider, 2019-11-28:
Creo que actualmente nadie trabaja a ese volumen, el grupo de primer es quizá el que más hasta y tenemos filamento que nos va a durar años

jgvictores, 2019-11-29:
Nos han llegado a pedir ciertos trabajos de elevado volumen y siempre se quedó en el aire debido al vacío en cuanto a estatutos.

GioSunrider, 2019-11-29:
Yo creo que ese tipo de trabajos deberian de aportar su propio plastico. Y se regirian por lo mismo motivo de los operadores de uso personal.

MrTommyGuns, 2019-11-29:
Estoy de acuerdo, deberían de llevar su material y/o darnos la pasta para pillar el material necesario para hacer lo que sea

jgvictores, 2019-11-29:
Lo que comentaba era respecto a un % adicional proporcional al utilizado, para compensar por el uso/desgaste/mantenimiento de las impresoras como tal.

GioSunrider, 2019-11-29:
Ahhh vale val, ya lo entinedo, y tambien me parece correcto. Ya que somos quienes las mantenemos y cuesta horas de trabajo tenerlas a punto.

closed 2023-09-28 by CarlosME17