asrob-uc3m/impresoras-asrob #153: Mejoras carro Spicy

JorFru, 2019-11-19:
Recomiendo añadir esta mejora a Spicy Caldito. https://www.thingiverse.com/thing:2326111
Yo tuve esta modificación en la mía y ayudaba mucho en caso de atasco, ya que puedes ver directamente el filamento pasar desde los engranajes del extrusor hasta que entra por el barrel.

Fácil de enebrar el filamento al cargarlo, fácil de desatascar. Y sobre todo que los nuevos pueden ver cómo es por dentro el asunto.
Respecto a la pregunta: ¿Hay que darle matarile al radiador?
La respuesta es: Sí, hay que cortarlo con una sierra. Es aluminio y se tardan tres segundos.
Este radiador es en un 80% inefectivo, y el ventilador un poco más de lo mismo.

Es una pieza muy fácil de imprimir y que aporta bastante creo yo. Y ya va preparando el camino a meterle un V6, que Spicy se lo merece.

MrTommyGuns, 2019-11-26:
Se realizó la impresión de la pieza y @fervh la instaló.
Tal vez sea preciso instalar unos tornillos más adecuados para que el ventilador se adapte apropiadamente.
Dejo la issue abierta para que @fervh la cierre si lo desea y pongamos alguna fotillo del resultado

luciagz02, 2019-11-28:
Ya se han instalado los tornillos adecuados y el ventilador encaja en su sitio.
[image]

closed 2019-11-28 by luciagz02