asrob-uc3m/impresoras-asrob #154: Impresora Ender PRO en usufructo

VentressAsajj, 2019-11-21:
Buenas, tengo una impresora Ender 3 PRO metida en la caja, recien comprada. Por problemas de sonido no la puedo montar en casa. He pensado que puedo dejarla en la asociación hasta que pueda llevármela a casa, cosa que no se cuándo ocurrirá. A cambio del espacio esa improsora está a disposición de la asociación.
El tema de filamento y compra de material en caso de avería deberíamos de dejarlo claro. 
No se si hay algo reglado.

GioSunrider, 2019-11-21:
Guiate de ejemplo del prestamo que realizo @JorFru con Spicy, aqui esta la [issue](https://github.com/asrob-uc3m/impresoras-asrob/issues/82) en la que establecio los terminos.

Por el tema del filamento note preocupes, se establece como cualquier impresora. de la asocicacion, es decir como se discute en #150

VentressAsajj, 2019-11-23:
@GioSunrider estoy de acuerdo con los términos de la impresora de @JorFru Lo mismo para la mía.

MrTommyGuns, 2020-02-13:
Hay que dotar a la impresora de un nombre official.

Que empiece la lluvia de ideas!!

MrTommyGuns, 2020-02-13:
Hoy se ha realizado el montaje de la impresora y se han impreso algunas piezas con resultados más que satisfactorios

closed 2023-09-28 by CarlosME17