asrob-uc3m/impresoras-asrob #155: Nombre para la nueva incorporacion

GioSunrider, 2019-11-28:
Como se establece en #152 tenemos que poner un nombre la nuestra pequeña, asi que que comiencen las propuestas.

David-Estevez, 2019-11-30:
Suegra

GioSunrider, 2019-12-07:
- Moeby Dick

- Moby Dick

MarioPrinter, 2019-12-07:
**NOMBRES PROPUESTOS:**
- Moeby Dick
- Moby Dick
- Suegra
- Flashy
- White
- Mr White
- Whity
- Little sister (y a la cube pro Big Daddy)
- Maincra

David-Estevez, 2019-12-07:
Blancanieves (o Snow White si os sentís internacionales)

Petikuis, 2019-12-12:
A mi me gusta la de Moeby Dick

GioSunrider, 2019-12-19:
Abierta votacion en https://forms.gle/h8whnzjjYJ5YPVbS8 os dejo una semanita.

JorFru, 2019-12-28:
¿Bueno cómo va esto?

GioSunrider, 2019-12-31:
Y cerramos el año con el nuevo nombre decidido. Moeby Dick ha sido el ganador. Procedere a los tramites para incorporarla en todos los lados.
[image]

closed 2020-02-13 by MrTommyGuns